dresden-elektronik/deconz-rest-plugin #251: deconz source

jonatanolofsson, 2017-11-03:
Pardon for not being strictly related to the rest-plugin, but instead to deconz.

Currently, programmatically interfacing with the conbee usb dongle requires not only the deconz software, but also this rest-plugin, which is (IMHO) a vastly over-complicated system which has hindered its adoption and interfacing with e.g. several of the smart home softwares available today.

Since I haven't found it anywhere, I would like to petition for considering releasing also the deconz software - or at least the conbee stick interface - as open source software to potentially harness the interest and power of us eager developers.

If this is already available, could you please point me in that direction?

Thank you for your work,

Best Regards,
Jonatan Olofsson

manup, 2017-11-03:
Beside REST API there is a non public documentation of the serial UART protocol between deCONZ and ConBee/RaspBee available on request (via invitation to private GitHub repo).

There are already developers from Open Hass and OpenHAB tinkering with it. I'm not aware how the progress is though.

You may refer to:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/158

https://community.openhab.org/t/eu-zigbee-dongle/31069/3

manup, 2017-11-22:
Close the issue for now.

closed 2017-11-22 by manup