dresden-elektronik/deconz-rest-plugin #254: Missing "pressed" event of Xiaomi Aqara Smart switch

jo-me, 2017-11-03:
Hi,
I've been playing with the Xiaomi / Aqara components and noticed that the [Aqara buttons](https://de.gearbest.com/access-control/pp_626695.html) behave differently than the older [Xiaomi smart buttons](https://www.gearbest.com/smart-light-bulb/pp_257679.html).

After adding them to deCONZ I watched the websocket events while pressing the buttons
The older Xiaomi Smart Switch triggers events (pressed / released) as expected:
```
{"e":"changed","id":"4","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2017-11-03T19:53:47"},"t":"event"}
{"e":"changed","id":"4","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-11-03T19:53:47"},"t":"event"}
```

The Aqara Switch, however, only triggers the "released" event:
```
{"e":"changed","id":"3","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-11-03T19:44:34"},"t":"event"}
```

Is this the normal behavior or bug?

Thanks
Jochen

manup, 2017-11-03:
Yep that's all the switch is sending, also it's problematic to press a button for several seconds (10?) because then the switch will leave the network.

HerrHofrat, 2017-11-07:
That's because the square aqara switch just sends the 1002 event. Unfortunately.

jo-me, 2017-11-07:
Thanks for clarifying this @manup and @HerrHofrat . I built a workaround for this in HomeAssistant that simulates a prior "pressed" event for 0.5s before forwarding this "released" event.

closed 2019-03-30 by stale[bot]