dresden-elektronik/deconz-rest-plugin #255: Beta app: allow specifying the http port

abmantis, 2017-11-03:
Currently my gateway is not using 80 nor 8080 ports, since those were already used on my raspberry pi. Can you allow specifying the port on the new beta app?

manup, 2017-11-03:
Yes, there is a workaround see:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/245

abmantis, 2017-11-03:
@manup oh great! Thanks.
By the way, I've tried adding a Xiaomi door sensor trough the beta, and it won't appear. Is that expected?

manup, 2017-11-03:
It should appear, however current uploaded version has bugs in the sensor wizards, which are fixed now. I'll upload the new version on weekend :)

abmantis, 2017-11-03:
That's great to hear!

abmantis, 2017-11-04:
@manup It appeared now. I think the issue is related to this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/250

After reseting the sensor multiple times in a row, it showed.