dresden-elektronik/deconz-rest-plugin #257: deconz not auto-discovered any more in Home Assistant

etique57, 2017-11-05:
Hello,

I can't really pinpoint when it happened (I'm transitionning from Osram to Deconz) but I remember the deconz app (or api) was automatically detected by Home Assistant.

It's not the case anymore (I'm currently on 2.04.85, hass.io being 0.57.1).

Any clue what could have gone wrong?

The beta app automatically detects the bridge, I cannot access the REST API, etc. so it's perfectly functional. Also, forcing the configuration of the hue bridge works properly.

Thanks!

Kane610, 2017-11-14:
Hey!

I'm not sure if this is allowed or not, but mods may remove it if it isn't.

I got a PR in the works for HASS for better integration with deconz. But I need more testers to verify support and functionality.

https://github.com/home-assistant/home-assistant/pull/10321

Feel free to try it out.

etique57, 2017-11-15:
Hi @Kane610 ,

More than willing to test, but you'll have to babysit me through the install of this wip component on my hassio instance...

staraxis, 2017-11-16:
I would be keen to test this as well, but like @etique57 above, I'll need to be walked through the installation of a dev component. Thanks for this, been hanging out for a deconz component since I got one!

Kane610, 2017-11-17:
Are you both using Hassio? Since I have nearly no experience with docker it might be complicated giving directions for that. If you're somewhat tech savvy and are a master in google fu, I think we can work something out :)

More or less what you need to do is to find the site-packages folder that contains a home-assistant folder that in turn keeps a components folder. When you've found the components folder it is just a matter of copy paste the deconz.py files in to corresponding sub folder, e.g. light/deconz.py to light/deconz.py. There are 4 files you need to move, components/deconz.py, components/light/deconz.py, components/binary_sensor/deconz.py and components/sensor/deconz.py

RudolfRendier, 2017-12-05:
I managed to set it up in a dockerized environment.
On top of the files mentioned here; others have been added in the meantime:

user@dockerhost:~/docker/hass/config$ ls custom_components/*
custom_components/deconz.py (this was called__init__.py, I think)
custom_components/services.yaml

custom_components/binary_sensor:
deconz.py 
custom_components/light:
deconz.py
custom_components/scene:
deconz.py 
custom_components/sensor:
deconz.py

On top of that you have to update discovery.py.
I bashed into my docker container and copied it from my config folder to homeassistant/compontents/

The component works really well. Please add/update instructions on how to install as custom_component to speed up user adoption and integration with upstream. 

----------------------------------------------------------------------------------------------------

I can also confirm the original issue; home assistant used to discover deconz as Hue, but not anymore.
I haven't updated home assistant in between, but I have been updating deCONZ.

Kane610, 2017-12-05:
Thanks for testing! 👍🏻
I will not spend time to document how to set it up as custom component as it is nearly ready to be integrated. 

Another user supported me to auto discover deconz so I have removed specific deconz support from the hue component.

RudolfRendier, 2017-12-11:
This seems to have been fixed after .94.
I'm currently on .97 and it's working again.

Still on home-assistant 0.57.3 (without the deconz-component by @Kane610) and it's detected as Hue.
Now I will re-install the custom_component again. :)

manup, 2018-01-30:
I close this for now since the case seams to be resolved.

closed 2018-01-30 by manup