dresden-elektronik/deconz-rest-plugin #258: Beta App: xiaomi senors in app.

wizkidorg, 2017-11-06:
I'm not using any lights, but rather a number of temperature/humidity sensors.  But the new Beta App doesn't display these devices at all even though the linux app show's them.  Previously in the webapp in modified the list of devices to show the temp sensors but am not able to modify the code for the new app since its fully hosted by you now.

Also in the Hue Bridge, the temperature sensors are also not being reported.

manup, 2017-11-06:
> I'm not using any lights, but rather a number of temperature/humidity sensors. But the new Beta App doesn't display these devices at all even though the linux app show's them. 

Which devices are they? The Phoscon app should show supported sensors (Philips, IKEA, Xiaomi, ...) with current values.

The sensors must have been added with a recent version of deCONZ (currently 2.04.86) to appear in the REST API.

> Previously in the webapp in modified the list of devices to show the temp sensors but am not able to modify the code for the new app since its fully hosted by you now.

Like the former webapp a copy of the Phoscon app will be included in deCONZ package and locally available once public release is ready.

> Also in the Hue Bridge, the temperature sensors are also not being reported.

What do you mean by hue bridge?

wizkidorg, 2017-11-06:
I'm not using any lights, but rather a number of temperature/humidity
sensors. But the new Beta App doesn't display these devices at all even
though the linux app show's them.

Which devices are they? The Phoscon app should show supported sensors
(Philips, IKEA, Xiaomi, ...) with current values.

The sensors must have been added with a recent version of deCONZ (currently
2.04.86) to appear in the REST API.

* I will delete all the devices and re-add them with the latest firmware
and see if that corrects it.  Will any other features (that you know of)
require delete the device and re-add to make them function?

Previously in the webapp in modified the list of devices to show the temp
sensors but am not able to modify the code for the new app since its fully
hosted by you now.

Like the former webapp a copy of the Phoscon app will be included in deCONZ
package and locally available once public release is ready.

* That is fine, was just curious if the new way of deploying the webapp was
going to be a "semi-cloud" method or not.

Also in the Hue Bridge, the temperature sensors are also not being reported.

What do you mean by hue bridge?

* I use smartthings as my primary bridge (it supports more then Zigbee, but
doesn't support these --Choose your 4-letter word substitute-- Xiaomi
device.) so I use the Hue Bridge (you open the deCONZ to your local
network, and smartthings reads the devices and makes them available to
itself, its builtin to the deCONZ main web application.) functionality in
the deCONZ system to make the devices available to SmartThings (deCONZ has
better zigbee support and doesn't become forgetful of the devices, e.g.
smartthings stops communicating with some devices after they stop
responding, but deCONZ will just keep on going and finally reconnect when
they wake up again.)

manup, 2017-11-06:
> * I will delete all the devices and re-add them with the latest firmware
and see if that corrects it.  Will any other features (that you know of)
require delete the device and re-add to make them function?

Make sure you add them via `Devices > Sensors > Add new sensor` (Phoscon app).

I guess when SmartThings doesn't show Xiaomi sensors provided by deCONZ REST API they are just not supported there — since original Hue bridge does not support them. This can only be added by SmartThings devs or by using these 'groovy' scripts, but I'm not sure how that works. Maybe people from the SmartThings forum can help here.

wizkidorg, 2017-11-06:
Ok finally got it to show the devices, the issue i think is the new app
only does lights initially, even though it never finds a light it gets
stuck in this loop.  I fully reloaded the webpage and now the menu shows
properly.

manup, 2017-11-06:
Yes it requires at least one lights or a group to get over the setup screens.
Interesting case, I'll forward that to the team.

rtenklooster, 2017-12-13:
The new wabapp is working great! My xiaomi motion sensors are listed fine. Adding new sensors is working great, Ikea bulbs are listed and working instant.

Only the xiaomi buttons aren't listed (yet) So keep on doing the good job :) thankks

closed 2018-09-17 by manup