dresden-elektronik/deconz-rest-plugin #259: FLS-PP3 sending unexpected responses

wvuyk, 2017-11-08:
We have been testing a FLS-PP3 ballast with the REST API and I think I am getting some weird effects, it feels like there is something moving to wrong areas here. We use the 2.04.86 version
When I send a hue command through the rest api, the light switches off after a second or two instead of changing color. The websocket response is sending a changed event for x or y where the value looks like being a hue value? Here some application logging of the communication, from sending command to response on the websocket:

`20:25:19.2378460 -- (BridgeApi.SendJsonCommand) SendJsonCommand: http://192.168.2.217:80/api/82D2F45A5A/lights/18/state 
{"on":true,"hue":28471,"alert":"none","effect":"none"}

20:25:19.2449940 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"18","r":"lights","state":{"x":17143},"t":"event"} new queue size is 0 on 819929332
20:25:19.2466580 -- (BridgeApi.CheckJsonResult) [{"success":{"/lights/18/state/on":true}},{"success":{"/lights/18/state/effect":"none"}},{"success":{"/lights/18/state/hue":28471}},{"success":{"/lights/18/state/alert":"none"}}]

20:25:19.2850130 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"18","r":"lights","state":{"y":30357},"t":"event"} new queue size is 0 on 819929332

20:25:22.7206500 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"18","r":"lights","state":{"on":false},"t":"event"} new queue size is 0 on 819929332`

Am I missing something, or is the rest api mixing values here?

Wim

manup, 2017-11-08:
It looks like the FLS is on color mode xy, I would recommend to add `sat` when using `hue` since these two are related.

```js
{"on":true, "hue": 28471, "sat":254, "alert":" none", "effect": "none"}
````

wvuyk, 2017-11-08:
I will have this tested, but still am surprised to see that the light just goes off because of this?

Wim

manup, 2017-11-08:
>I will have this tested, but still am surprised to see that the light just goes off because of this?

Yep this might be a firmware bug, I need to check what happens there.

wvuyk, 2017-11-09:
Did some more testing here on getting the hue value to the FLS-PP3 and we think the issue is with sending the alert command? 
This command:
`ph_put /lights/18/state '{"on":true,"hue":35300,"alert":"none"}`
puts the light off, where this command
`ph_put /lights/18/state '{"on":true,"hue":35300}` works as expected. Still then I wonder why the high xy values are reported. Why doesn't it switch to hs mode automatically then?

Wim

wvuyk, 2017-11-09:
There is somethng off with the xy values, part of my finding earlier. Tonight I am testing a bit more with the xy values and I keep geting high values back from the websocket. Values on the rest api seem to be correct...

here is some tracing I did on sending a command to a IKEA RGB and receiving the reply from the websocket. It seems to be illogical that x and y values are send in two replies?

`2:52:42.4000757 -- (BridgeApi.SendJsonCommand) SendJsonCommand: http://192.168.192.11:80/api/F8F24CD00B/lights/16/state 
{"on":true,"xy":[0.2011,0.1293]}
22:52:42.4070477 -- (BridgeApi.CheckJsonResult) [{"success":{"/lights/16/state/on":true}},{"success":{"/lights/16/state/xy":[0.2011,0.1293]}}]

22:52:42.4080431 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"16","r":"lights","state":{"x":13127},"t":"event"} new queue size is 0 on 00212EFFFF0102D3
22:52:42.4505539 -- (WebCommander.HandleWebQueue) Dequeue command: json {"e":"changed","id":"16","r":"lights","state":{"y":8440},"t":"event"} new queue size is 0 on 00212EFFFF0102D3
`
so I am getting x:13127 and y:8440 back on a xy command? What are these high values?
when I check the REST API I am getting values that are expected:

`{
    "ctmax": 454,
    "ctmin": 250,
    "etag": "bf8ef5bab61e91070c7a2f37b3c5f39c",
    "hascolor": true,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 CWS opal 600lm",
    "name": "Tradfri Color",
    "state": {
        "alert": "none",
        "bri": 64,
        "colormode": "xy",
        "effect": "none",
        "hue": 0,
        "on": true,
        "reachable": true,
        "sat": 0,
        "xy": [
            **0.201091,
            0.129291**
        ]
    },
    "swversion": "1.3.002",
    "type": "Color light",
    "uniqueid": "00:0b:57:ff:fe:a7:e9:46-01"
}`

ebaauw, 2017-11-09:
The `x` and `y` values in the websocket notification are the raw values of the corresponding ZigBee attributes; the `xy` values in the REST API are normalised to the range 0.0...1.0.  deCONZ still uses a different translation between these than the Hue bridge, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/195#issuecomment-334960340.

wvuyk, 2017-11-09:
I understand it, but why using a different appraoch on the websocket compared to the REST API?
Either use the raw values and use it everywhere or use the normalised values, not all. That only complicates things for others. Which one, does not matter to me, but choose ;-)

manup, 2017-11-10:
> I understand it, but why using a different appraoch on the websocket compared to the REST API?

This will be fixed in the next versions, together with the number differences which Erik mentioned. The current state is due the separated handling of the x/y values on ZigBee level.

closed 2018-09-17 by manup