dresden-elektronik/deconz-rest-plugin #262: Windows ports

chr55, 2017-11-09:
I'm using Windows version 2.04.86 installed on the same Windows 10 machine that has HomeSeer. HomeSeer is using port 80 for it's web server.  When I start deCONZ.exe after HomeSeer has launched everything is fine. My issue is when I launch deCONZ.exe after logging in using the Windows Task Scheduler, say after patches have been applied. I need to be able to specify exact ports for the web interface and the api since deCONZ loads first. I have tried a delay and that doesn't help either. 

I have tried passing the --http-port and the --api-port parameters on the Windows Task scheduler item but they do not seem to work for me.  I also have found a configuration file here, C:\Users\HomeSeer\AppData\Local\dresden-elektronik\deCONZ\config.ini which also does not seem to be read when I launch it.   The executable was installed here C:\Program Files (x86)\deCONZ\bin\deCONZ.exe

It would be very helpful if the two ports could be updated in the deCONZ interface or if someone could explain to be how I can reliably get a consistent set of ports. 

Thank you!

manup, 2017-11-09:
The `--http-port=<port>` command line parameter should work, at work I'm using this in a modified desktop shortcut to open deCONZ with various parameters.

chr55, 2017-11-10:
Thank you for the tip!  I have it working. Who would have thought that using Windows Task scheduler would not work. I had Windows Task scheduler, run on login, with the parameters etc. Not working. 

Your procedure with the shortcut is working for me. I used the procedure outlined here to get the shortcut into the startup folder.

https://grok.lsu.edu/article.aspx?articleid=19301

For the target I put "C:\Program Files (x86)\deCONZ\bin\deCONZ.exe" --http-port=8080

I rebooted the Windows machine and deCONZ started on port 8080. Awesome!

Again, Windows Task scheduler not working, I tried again.

Another thing in case this helps anyone else, clicking the deCONZ button "WebApp" brings up a browser to http://IP-OF-THE-MACHINE:8080 which does not work on the deCONZ server itself. It works fine on another client.  On the server it needs to be http://localhost:8080

thanks,

closed 2017-11-10 by chr55