dresden-elektronik/deconz-rest-plugin #263: Xiaomi Aqara flood

Kohle81, 2017-11-10:
I want to include this sensor with deCONZ to Homekit with homebridge-hue plugin.

Does it work or what can I do?

Greetings

manup, 2017-11-10:
To start with it would be nice to have a screenshot from the sensor with expanded cluster list, for that click on second circle on the sensor node.

Also the model id which the sensor reports in the `Basic Cluster` in the cluster info panel.
If it is empty you may need to rejoin the sensor whilte the network is open, for Xiaomi sensors there should be a button on the device.

Kohle81, 2017-11-10:
Ok, I will send a photo next days. Actually I have this picture on my smartphone from the sensor.
[image]

manup, 2017-11-10:
That only shows the mac address (00158d...), for the integration the other infos are needed :)

oleg77s, 2017-11-10:
In my system:
[image]

ebaauw, 2017-11-10:
Hm, that doesn’t look good: there should be another cluster where the sensor reports its state.  Could you try and repair the sensor while regularly pressing the sensor’s button (or hole), to force it to remain awake during pairing?

oleg77s, 2017-11-10:
yes, i try, but there is only basic cluster

Kohle81, 2017-11-21:
I have only the same fluster Info. There is no more Information.......

pietervandersluis, 2017-11-23:
I had the same issue with a aqara wall switch. Have you tried re-pairing the device?

Kohle81, 2017-11-24:
I have re-paired two times. Always the same...

oleg77s, 2017-11-24:
Xiaomi Smoke sensor: only basic cluster too :(

ebaauw, 2017-11-25:
See #152 for the smoke detector.

Reading back that issue, maybe Xiaomi don’t report the cluster in the device description (so deCONZ GUI won’t show it), but does send commands from the cluster anyways.  Wouldn’t be the first time they’re creative with the standard.  The bad news: we need to sniff the ZigBee traffic to figure out the flood sensor.

QempZor, 2018-01-24:
Sad news that it is still not working :
Got my flood sensor today, hoping it was already recognized by homebridge-hue. @ebaauw any news yet?

ebaauw, 2018-01-24:
homebridge-hue can only expose devices that are supported through the deCONZ REST API.

To support the flood sensor in the deCONZ REST API, we need to know how the flood sensor communicates.  For standard ZigBee devices, typically the screenshots from the GUI are enough to figure that out.  For more exotic ZigBee devices, like the Xioami, we need sone-one to sniff the ZigBee traffic.

QempZor, 2018-01-24:
I already thought that. To bad I can't sniff it myself, don't got a 802.15.4 sniffer.
I've someone is intressted to sniff it, I can provide a flood sensor (dutch only, because otherwise it will be to expensive to send haha)

ebaauw, 2018-01-24:
I’ll be more than happy to have a look at it, if I can borrow one.

QempZor, 2018-01-24:
That would be awesome! Just send you a message (facebook).
Couldn't find your email/private message.

ebaauw, 2018-01-27:
Thanks @QempZor, I received the sensor.  Took me a while to decipher the booklet and figure out the sensor can be squeezed, but after that, I managed to pair it to deCONZ.

Indeed, it only advertises a _Basic_ cluster on endpoint 0x01.  With some effort, it responds to reading the attributes, while squeezing it:
[image]

When squeezed, it sends a manufacturer-specific report attributes command (0x0a) for manufacturer code 0x115f.  It seems to contain attribute 0x0005, `lumi.sensor_wleak.aq1` and attribute 0xff01, type 0x042 (character string) of length 34.  Wireshark reports a malformed packet.  This is typical Xiaomi behaviour - the battery status is probably encoded in there somewhere.

When the contacts are shortcut (simulating a leak detection), it sends a regular IAS Zone Status Change Notification (cluster 0x0500, command 0x00) with ZoneStatus 0x0001 (alarm 1 set) and ZoneID 0xff.  When the shortcut is removed, after a while is sends another IAS Zone Status Change with ZoneStatus 0x0000.

I haven't dealt with IAS zone devices before, but it should be pretty straightforward to support the leak sensor in deCONZ (and, after that, in homebridge-hue).

ebaauw, 2018-01-27:
The good news: I got the flood sensor to work with deCONZ.  `state.presence` changes when I shortcut the contacts, and when I remove the shortcut.  The resource looks like:
```json
{
  "config": {
    "duration": 0,
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "5562c5bf35401b7b29308c21a08a5d51",
  "manufacturername": "LUMI",
  "modelid": "lumi.sensor_wleak.aq1",
  "name": "Test",
  "state": {
    "lastupdated": "2018-01-27T19:56:45",
    "presence": false
  },
  "swversion": "3000-0001",
  "type": "ZHAPresence",
  "uniqueid": "00:15:8d:00:01:bb:61:59-01-0500"
}
```
The bad news: the REST API plugin still doesn't create the sensor resource on pairing.  I added the sensor manually to the database, including the _IAS Zone_ cluster in the fingerprint.  As the sensor doesn't advertise the _IAS Zone_ cluster, I think somewhere during discovery, we should insert it into the fingerprint, based on the sensor model.  I don't know where in the code to do that.  @manup, can you help me out?

Did we decide yet on an attribute for the IAS zone type?  I think we should use 0x002a (Water sensor) for the flood sensor, but, of course, the sensor doesn't report this attribute.

QempZor, 2018-01-29:
That sounds hopefull. Nice to hear you already so far, well done!

I think we should use ZHAPresence for this type. Like said by @rtenklooster in #152,
We could use ZHAAlarm for Floodsensors, Smoke decters, Gas detectors, however if you already have implemented ZHAPresence, then we should you use that. Detecting the presence of smoke, gas and water seems very legit to me.

Got my Smoke detector this weekend, wich also got the same issues as #152.

Webserve, 2018-02-01:
Thanks already @ebaauw just received my aqara water sensor. Hope that you can make it work :)

ebaauw, 2018-02-01:
@manup I understand you're busy, but I would appreciate if you can have a look at this.  Where in the code would I "insert" an _IAS Zone_ cluster to the leak sensor?

manup, 2018-02-01:
Hi @ebaauw sorry I've missed the above.

Normally deCONZ should add the missing server clusters on reception of related commands if the profileId and clusterId is found in the ZCLDB XML file. It should then be visible in the GUI endpoint list.

But it might be easier to add the sensor by just looking at the basic clusters modelid and if it is `lumi.sensor_wleak.aq1`  exec `fpPresenceSensor.inClusters.push_back(IAS_ZONE_CLUSTER_ID);`

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L2582

ebaauw, 2018-02-01:
Thanks, @manup, that did the trick.  I've created a PR.

@Kohle81, @QempZor, @Webserve, @oleg77s, To pair the flood, open the network _from the web app_, squeeze the sensor for 10 seconds to reset it, and then briefly squeeze it.  It appears in the GUI and the REST sensor resource is created straight away.    homebridge-hue already supports the flood, see https://github.com/ebaauw/homebridge-hue/issues/205.

> Normally deCONZ should add the missing server clusters on reception of related commands if the profileId and clusterId is found in the ZCLDB XML file. It should then be visible in the GUI endpoint list.

It's not, then again Xiaomi != normal, I suppose.  However, it does send a regular IAS Zone Status Change Notification (cluster 0x0500, command 0x00) and it's on profile ZHA.  Maybe the device ID also needs to be specified in general.xml?

> I think we should use ZHAPresence for this type. Like said by @rtenklooster in #152,
We could use ZHAAlarm for Floodsensors, Smoke decters, Gas detectors

I think I would favour ZHAAlarm (and a matching CLIPAlarm).  Obviously, ZHAPresence is less work, but I've seen multiple requests for an IAS Zone type attribute.  Also, with ZHAPresence, you get a `config.duration` with a default value of  60 for free, which is totally not needed, as the sensor sends a new message to turn off the alarm, and probably not wanted, when you want to know when the sensor ends the alarm.

I'm expecting to receive a bunch of loaner Heiman gadgets this weekend (smoke, gas, CO, temperature, door, motion, and leak sensors; power plug; and siren).  I'd be willing to give ZHAAlarm a go.

manup, 2018-02-02:
Thanks @ebaauw PR is merged now.

I also favour a ZHAAlarm, the related REST attribute (alarm?!) should allow various values. Maybe a 32-bit bitmap value is suitable to support multiple flags?

>I'm expecting to receive a bunch of loaner Heiman gadgets this weekend (smoke, gas, CO, temperature, door, motion, and leak sensors; power plug; and siren). I'd be willing to give ZHAAlarm a go.

Oh my, that's a lot stuff to play with :)

ldplusse, 2018-02-03:
The change to get the Xiaomi water sensor to work, is that going to be in the next DeCONZ release.
Not sure how to understand ebaauw's message.

ebaauw, 2018-02-03:
@ldplusse, if you compile the REST API plugin from the latest commit, you can use the leak sensor straight away.  Otherwise, you'll have to wait for the next deCONZ release.

Webserve, 2018-02-04:
Thanks @ebaauw its working. The sensor is not showing up in the GUI but with the api i can check the status.

ebaauw, 2018-02-04:
@Webserve,

> How can i disable the duration setting.

Through the REST API, issue a PUT of /sensors/xx/config with a body of `{"duration":0}`.

> The sensor is not showing up in the GUI

That's odd.  Are you sure?  Did you double-check the _Node List_ panel?  Sometimes newly discovered nodes are hidden behind existing ones.

ebaauw, 2018-02-18:
As of v2.05.04, the leak is exposed as ZHAWater, see https://github.com/ebaauw/homebridge-hue/issues/205#issuecomment-366534143.

QempZor, 2018-02-18:
Thats very nice! Great job @ebaauw time to order some more soon :)

martikainen87, 2018-02-24:
How do you get them to connect with deconz? I'm running 2.05.05 and tried a lot of combinations

- 5 second hold
- 1 time click
- 3 quick clicks
- 3 slow slicks

and then all of the above and at the same time putting the sensor on a wet paper towel

manup, 2018-02-24:
Have you started sensor search in Phoscon App?

 *Menu > Devices > Sensors > Add New Sensor > Other*

ebaauw, 2018-02-24:
I had hard time figuring out that the sensor could be "squeezed".  Then I tried all sorts of things: pressing 10x, holding for 10 seconds, removing the battery, until it appeared in the deCONZ GUI.  I'm not sure what the magical combination is...

Once the sensor is recognised in the GUI, keep on pressing/releasing while reading the _Basic_ cluster.

martikainen87, 2018-02-24:
Was able to get 1 out of 2 flood sensors working, and I didn't understand what I did to get it to work either :D 
The second one I've tried everything I can find, some smartthings people saying that you can hold it for 1+ minutes and the device will start reseting. The problem with mine is that it won't change the behaviour of the led

holding it results in

- solid blue light for 3 seconds
- led off for half a second
- solid blue light for 3 seconds
- led off for half a second
and it keeps repeating that pattern for over 2 minutes. Replaced the battery, tried triggering it with a wet towel.
Now my thumbs are to sore to keep trying, and I can't add it to my xiaomi hub neither, so something is up with it, gonna leave it without the battery for a while and see if it helps

Here's the thread from smarthings forum with a few tricks that anyone who is facing problems can use.
https://community.smartthings.com/t/xiaomi-aqara-water-sensor/97584/60

oleg77s, 2018-02-24:
Aqara flood sensor is working now :)
DeConz and Homebridge-hue -> latest
[image]
[image]
[image]

LordVoodoo, 2018-02-28:
Hello everyone,

I have a little trouble with the device. I have updated the Deconz package to the latest beta (2.5.0.7) and in the Phoscon App when joining the flood sensor, the window colours green shortly, but afterwards the device is not listed/displayed in Phoscon App.

If I search for sensors in my Smart Home based on FHEM I see the sensor, but I do not get any readings. Even after watering the sensor no event...

...any idea, where to look for my mistake?

Thanks in advance.
LordVoodoo

martikainen87, 2018-03-01:
The listing doesn't work yet, check it by using rest api or in HA if you are pushing data to it (upgrade to 0.64.1 version since the support for flood sensor is new)

LordVoodoo, 2018-03-01:
With ‘HA’ you mean HA-Bridge?

martikainen87, 2018-03-01:
HomeAssistant together with the Deconz component (working like a bridge between deconz and HA)

Alexxander0, 2018-09-29:
I've been able to pair one leakage sensor. It worked fine via API.
Today i updated the software to  2.05.39 / 22.9.2018 and iam not able to pair a new leakage sensor. Not 100% sure how i paired the first one. But there is no new sensor in the API.
Any help? How do i debug? Dont have the GUI installed

Mevel, 2018-12-16:
Same here: software to 2.05.39 / 22.9.2018

It gets Paired and shows up in the `/api` Json call but it wont show in the Phoscon webapp

manup, 2018-12-16:
Water sensor should now pair and be shown in 2.05.51

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_51

thecem, 2019-01-14:
> Paired one Xiaomi water sensor yesterday with .54 without issues. Started Search for Sensors -> Others, pressed the Sensor for about 10 secs and then continued pressing every 3-5 secs until the sensor was seen properly in deCONZ.

hmm, normaly you have to press for 5 Sec and then it connects (start blinking, release button, 3sec light pause, 3 times blinking, 2sec,  connected), (like Aqara Home App describes to connect to the aqara hub), and this is how it works after downgrade to .51 after upgrade to .54 same problem here and not able to connect.

Webserve, 2019-01-14:
.52 is repaired my water sensor without any issue. 

Just in de gui other sensor -) press button until flashing light -) then release -) press once more and the sensor was connected

manup, 2019-01-16:
Please try with 2.05.55, there were many fixes for related issues.

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_55

closed 2019-05-23 by stale[bot]

flopp999, 2020-02-18:
I can add this without problem and it works great to detect water and it shows up in Domoticz :)
I do see temperature in deconz, but it doesn't show up in Domoticz. Is temperature just an extra feature that Aqara doesn't say is a trusted value?