dresden-elektronik/deconz-rest-plugin #265: Unreliable light switching

djwlindenaar, 2017-11-11:
Hi,
I'm having an issue with a simple setup of a switch and light. I'm using the "new type" xiaomi light switch (WXKG03LM) and a Xiaomi Aqara smart socket. Since the button only reports a button press, I've created a simple set of 2 rules to respond to any report update and in response switch the light on if it was of and vice versa.

The issue seems to be in how the smart socket gets switched through the rules or groups. When I press the button, the socket stops switching and I can trigger the same behavior in the web interface. Whenever in the deCONZ GUI I press the off or on button in the sockets 0006 cluster, the sockets switches happily and it seems after that the web interface as well as the button have an effect again.

When the socket does not respond, the websocket messages still happen even though (in the second event) the socket was left in the off state:
":"2017-11-11T08:15:09"},"t":"event"}
< {"e":"changed","id":"1","r":"lights","state":{"on":false},"t":"event"}
< {"e":"changed","id":"0","r":"groups","state":{"any_on":false},"t":"event"}
< {"e":"changed","id":"1","r":"groups","state":{"any_on":false},"t":"event"}
< {"e":"changed","id":"3","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2017-11-11T08:15:10"},"t":"event"}
< {"e":"changed","id":"1","r":"lights","state":{"on":true},"t":"event"}
< {"e":"changed","id":"0","r":"groups","state":{"any_on":true},"t":"event"}
< {"e":"changed","id":"1","r":"groups","state":{"any_on":true},"t":"event"}

Corresponding log:
the socket toggle at 8:15:09 works, the one at 8:15:10 not. B72 is the socket, F84 is the button

08:15:09:033 ZCL attribute report 0x00158D000183CF84 for cluster 0x0006
08:15:09:034 Node data 0x00158d000183cf84 profileId: 0x0104, clusterId: 0x0006
08:15:09:034 update ZCL value 0x0006/0x0000 for 0x00158D000183CF84 after 6 s
08:15:09:039 discard sensor state push for state/lastupdated (already pushed)
08:15:09:039 rule event: /sensors/3 state/lastupdated num (0 -> 0)
08:15:09:039 trigger rule 4 - test2
08:15:09:040 add task 395634 type 14 to group 0x0001 cluster 0x0006 req.id 178
08:15:09:040 trigger rule events took 1 ms
08:15:09:255 ZCL attribute report 0x00158D0001701B72 for cluster 0x0006
08:15:09:256 Node data 0x00158d0001701b72 profileId: 0x0104, clusterId: 0x0006
08:15:09:259 Node 0x00158D0001701B72 is known by 1 neighbors, last seen 0 s
08:15:09:268 	0x00158D0001701B72 force poll
08:15:09:268 Erase task req-id: 178, type: 14 zcl seqno: 2 send time 0, profileId: 0x0104, clusterId: 0x0006
08:15:09:737 Node 0x00158D000170CC76 is known by 1 neighbors, last seen 50 s
08:15:10:216 Node 0x00158D000183CF84 is known by 1 neighbors, last seen 1 s
08:15:10:697 Node 0x00158D00018B4B00 is known by 0 neighbors, last seen 85633 s
08:15:10:752 ZCL attribute report 0x00158D000183CF84 for cluster 0x0006
08:15:10:752 Node data 0x00158d000183cf84 profileId: 0x0104, clusterId: 0x0006
08:15:10:752 update ZCL value 0x0006/0x0000 for 0x00158D000183CF84 after 1 s
08:15:10:767 discard sensor state push for state/lastupdated (already pushed)
08:15:10:768 rule event: /sensors/3 state/lastupdated num (0 -> 0)
08:15:10:769 trigger rule 3 - test2
08:15:10:770 add task 395646 type 14 to group 0x0001 cluster 0x0006 req.id 198
08:15:10:771 trigger rule events took 2 ms
08:15:11:176 Node 0x00158D0001701B72 is known by 1 neighbors, last seen 1 s
08:15:11:595 	0x00158D0001701B72 force poll
08:15:11:596 Erase task req-id: 198, type: 14 zcl seqno: 3 send time 1, profileId: 0x0104, clusterId: 0x0006
08:15:11:656 Node 0x00158D000170CC76 is known by 1 neighbors, last seen 52 s
08:15:12:137 Node 0x00158D000183CF84 is known by 1 neighbors, last seen 1 s
08:15:12:616 Node 0x00158D00018B4B00 is known by 0 neighbors, last seen 85635 s
08:15:13:096 Node 0x00158D0001701B72 is known by 1 neighbors, last seen 3 s
08:15:13:576 Node 0x00158D000170CC76 is known by 1 neighbors, last seen 54 s
08:15:14:057 Node 0x00158D000183CF84 is known by 1 neighbors, last seen 3 s
08:15:14:537 Node 0x00158D00018B4B00 is known by 0 neighbors, last seen 85637 s

Any clue on what's happening here and how to solve? (unreliable lights are a bit of a no-go :)

Best regards
Daniel

manup, 2017-11-11:
Can you please provide the related rules?
Does the socket respond reliable when you turn it on/off via the buttons of the group in the webapp?

ebaauw, 2017-11-11:
You need to remove the socket from the switch group.  Otherwise the socket gets switched off by the rule, and on by the group at the same time as result of the second press.

djwlindenaar, 2017-11-11:
The webapp group is equally unreliable. In fact it's correlated. When webapp does not switch the light, the button/rules don't work either...

Here's the rules:
    "rules":
    {
        "3":
        {
            "actions":
            [
                {
                    "address": "/groups/1/action",
                    "body":
                    {
                        "on": true
                    },
                    "method": "PUT"
                }
            ],
            "conditions":
            [
                {
                    "address": "/sensors/3/state/lastupdated",
                    "operator": "dx"
                },
                {
                    "address": "/groups/1/state/any_on",
                    "operator": "eq",
                    "value": "false"
                }
            ],
            "created": "2017-11-10T09:22:58",
            "etag": "9b2b14e12607a515f022bebe64120384",
            "lasttriggered": "2017-11-11T12:09:02",
            "name": "test2",
            "owner": "747C417170",
            "periodic": 0,
            "status": "enabled",
            "timestriggered": 93
        },
        "4":
        {
            "actions":
            [
                {
                    "address": "/groups/1/action",
                    "body":
                    {
                        "on": false
                    },
                    "method": "PUT"
                }
            ],
            "conditions":
            [
                {
                    "address": "/sensors/3/state/lastupdated",
                    "operator": "dx"
                },
                {
                    "address": "/groups/1/state/any_on",
                    "operator": "eq",
                    "value": "true"
                }
            ],
            "created": "2017-11-10T09:23:12",
            "etag": "095e0ee6656f355426f0a0fc943455fe",
            "lasttriggered": "2017-11-11T12:08:53",
            "name": "test2",
            "owner": "747C417170",
            "periodic": 0,
            "status": "enabled",
            "timestriggered": 88
        }
    }

@ebaauw, I'm not sure what you mean by the "switch group". This is what I've got in REST:
    "groups":
    {
        "1":
        {
            "action":
            {
                "bri": 127,
                "colormode": "hs",
                "ct": 0,
                "effect": "none",
                "hue": 0,
                "on": true,
                "sat": 127,
                "xy":
                [
                    0,
                    0
                ]
            },
            "class": "Other",
            "devicemembership":
            [
            ],
            "etag": "6aa96bd7fad57799ca667b50be4b0c25",
            "hidden": false,
            "id": "1",
            "lights":
            [
                "1"
            ],
            "lightsequence":
            [
                "1"
            ],
            "multideviceids":
            [
            ],
            "name": "Group 1",
            "scenes":
            [
            ],
            "state":
            {
                "any_on": true
            },
            "type": "LightGroup"
        }
    },

manup, 2017-11-11:
The Xiaomi switches don't have a own group, they only send unicast to the gateway which should handle the switching.

It looks like the socket doesn't support group commands very well, as a workaround you may add actions to PUT `on` state (/lights/1/state) of the socket. This will trigger unicast commands to the socket which it seems to support better.

djwlindenaar, 2017-11-11:
OK, I'll try that. So far it seems to work, but i'll go ahead and stress test it for a while. 

Thanks!

djwlindenaar, 2017-11-12:
OK, reliability is good now, but... I've just closed and restarted deCONZ and now the rules don't trigger anymore. Not sure why not, is there anything I need to do to have the rules trigger again after a restart of deCONZ?

Rules are now:

    {
        "3":
        {
            "actions":
            [
                {
                    "address": "/lights/1/state",
                    "body":
                    {
                        "on": true
                    },
                    "method": "PUT"
                }
            ],
            "conditions":
            [
                {
                    "address": "/sensors/3/state/lastupdated",
                    "operator": "dx"
                },
                {
                    "address": "/lights/1/state/on",
                    "operator": "eq",
                    "value": "false"
                }
            ],
            "created": "2017-11-10T09:22:58",
            "etag": "9926aa3493c46a7efcf4ebed58e9fae5",
            "lasttriggered": "none",
            "name": "test2",
            "owner": "747C417170",
            "periodic": 0,
            "status": "enabled",
            "timestriggered": 114
        },
        "4":
        {
            "actions":
            [
                {
                    "address": "/lights/1/state",
                    "body":
                    {
                        "on": false
                    },
                    "method": "PUT"
                }
            ],
            "conditions":
            [
                {
                    "address": "/sensors/3/state/lastupdated",
                    "operator": "dx"
                },
                {
                    "address": "/lights/1/state/on",
                    "operator": "eq",
                    "value": "true"
                }
            ],
            "created": "2017-11-10T09:23:12",
            "etag": "7cf5533e4e4398518420b99093721bce",
            "lasttriggered": "none",
            "name": "test2",
            "owner": "747C417170",
            "periodic": 0,
            "status": "enabled",
            "timestriggered": 107
        }
    }

ebaauw, 2017-11-12:
Does deCONZ already support conditions on light state?  You might need a condition on group state.any_on and an action for light state.on.

djwlindenaar, 2017-11-12:
Looks like the condition works, but somehow needs to be primed... I've just found the following.
When I restart deCONZ, the rules don't fire. When I issue a PUT request through the REST api on the lights/1/state "on":true (or false, doesn't matter), the rules start working again.

It looks like somehow the condition on light state.on neither equals true nor false after startup... I could work around this by changing the rule set, but it seems undesirable behavior.

manup, 2017-11-12:
Problem here is that after restart the `group.any_on` takes some time to be figured out (polling or attribute reporting). This may take a few minutes.

ebaauw, 2017-11-12:
> Looks like the condition works

That's good news!

> It looks like somehow the condition on light state.on neither equals true nor false after startup

Funny.  I noticed something similar on conditions on CLIP sensors state.flag or state.status.  After startup, deCONZ wouldn't fire any rules, until I changed the value once.  Doesn't seem to happen all the time, though.

I hacked together a set of rules that allow me to do some actions on deCONZ startup time.  I use this to set a CLIPGenericFlag recording the startup time, and to turn off all lights (in case the restart is due to a power failure recovery).  As a workaround, you could try and use a similar scheme to "prime" the light.

The following two rules set a flag, `/sensors/103`, for Night Mode, but one them also fires when deCONZ starts:
```json
{
  "actions": [
    {
      "address": "/sensors/103/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T23:30:00/T07:00:00"
    }
  ],
  "name": "Night On",
  "status": "enabled"
}
```
```json
{
  "actions": [
    {
      "address": "/sensors/103/state",
      "body": {
        "flag": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/config/localtime",
      "operator": "in",
      "value": "T07:00:00/T23:30:00"
    }
  ],
  "name": "Night Off",
  "status": "enabled"
}
```
The next rule sets a flag, `/sensors/107`, for Boot Time and turns off the lights:
```json
{
  "actions": [
    {
      "address": "/sensors/107/state",
      "body": {
        "flag": true
      },
      "method": "PUT"
    },
    {
      "address": "/groups/0/action",
      "body": {
        "on": false
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/103/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/107/state/flag",
      "operator": "eq",
      "value": "false"
    }
  ],
  "name": "Boot Time",
  "status": "enabled"
}
```

djwlindenaar, 2017-11-12:
OK. Nice workaround, I can use that.

I'll check if the light reports its status at some point, but I've got some doubts if it will as this is the xiaomi plug... It also does not seem to report back if the light is switched on but the pushbutton on the socket, I think.

I feel that having any undefined status is not great although I can see that same startup values might be difficult to define. Or should we have an explicit undef status or something like that?

Best regards
Daniel

djwlindenaar, 2017-11-18:
Yep. Problem solved. Thanks for the help

closed 2017-11-18 by djwlindenaar