dresden-elektronik/deconz-rest-plugin #266: Zigbee Light Link

mtornblad, 2017-11-11:
I have been able to join a Hue Bridge v2.1 with latest firmware into an ZLL network created by Ikea Gateway. It is working but is a bit unstable in the different way lamps are working, some lamps can only be reached through Ikea App, some from Hue App, and some can be controlled from both apps. I would like to join deConz into this network to find out more what is happening. Is that possible? I have found this links but none of them seems to work:
#21 ZLL Commissioning
[Quick start guide deCONZ ZigBee Light Link](https://www.google.se/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&ved=0ahUKEwifzvXPxbfXAhWIJZoKHdwfD8UQFgg4MAI&url=https%3A%2F%2Fwww.dresden-elektronik.de%2Ffunktechnik%2Fproducts%2Fsoftware%2Fpc%2Fdeconz%2F%3FeID%3Ddam_frontend_push%26docID%3D1081&usg=AOvVaw3nepQdmrHbTkmlte_G4M3f)

Since there are no clear guide on how to join Hue to an existing network, this is what I found out after a lot of reading and trial & errors!

1. Set up Ikea Gateway the "standard" way together with one Ikea remote.
2. Connect Hue to Network and power it up
3. Press big button on top and keep it down
4. In deConz under "Reset Devices via touchlink, press "Scan.."
5. To if more than one deivce are found, "Blink" should flash blue ring on Hue Bridge.
6. With button still pressed do a "reset" in deConz.
7. Release Hue button and wait a while for Hue bridge to reboot and come up as Factory new 
8. (If you like you can verify that it is in Factory new with new scan as above, panid should be FFFE)
9. Press big button on top of Hue bridge again and keep it down
10. Touchlink the configured Ikea remote from step 1 by holding it close to Hue Bridge and press link button on the back. When red light on Ikea remote stops flickering you can relase.
11. To verify that Hue has joined the same network as Ikea Gateway, keep lamp configured in step 1, Ikea GW and Hue bridge near deConz and do a new "Touchlink scan".

//Mattias

ebaauw, 2017-11-18:
Interesting, it would seem that the Hue bridge still supports being touch linked into an existing network, see #117.  And you don't need a Living Colors remote!  Does the Hue bridge show as a router (yellow) or as a gateway (blue) in the deCONZ GUI?

This setup might actually be preferred over configuring deCONZ as a router on the Hue bridge network, see #21.  Still, the usual caveats would apply: typically devices can only report to one gateway/bridge and you don't want multiple gateways/bridges polling the same lights.

ebaauw, 2017-11-18:
OK, I can factory-new reset the v2 Hue bridge when holding the link button through deCONZ touchlink as well as through the Hue dimmer switch (holding On and Off for 10 seconds).  It seems the link button doesn't light up when the bridge is in factory new state.  deCONZ touchlink confirms its on panid 0xfffe.

However, I seem to be stuck here.  I have two Trådfri remotes and one Trådfri motion sensor linked to deCONZ, but their light doesn't come on when holding the link button on the back for 10 seconds, while keeping them next to the Hue bridge (still in factory new state) while holding the bridge's link button.  I double-checked that the remote will pair to an IKEA bulb while holding the link button on the back, joining the bulb to the deCONZ network.

manup, 2017-11-18:
During Touchlink various messages are exchanged which also contain information about the device type, I reckon the IKEA remotes will only proceed then the other device matches specific attributes (lights for example).

In the sniffer you should see the scan request and responses, which give more insight.

mtornblad, 2017-11-18:
The remote i used in both steps came from an existing Ikea network, and was initiated with touchlink! I think i have read somewhere that it is not allowed to touchlink a remote that are initiated in a HA network with trust center! If you reset the remote and one bulb and let them create a new "non trustcenter" network, can that remote then be used in the last step?

ebaauw, 2017-11-18:
> it is not allowed to touchlink a remote that are initiated in a HA network with trust center

Then in shouldn't have touch linked the light, shouldn't it?  When the Hue dimmer switch is paired with deCONZ, we set a manufacturer specific attribute to disable touch linking, just as the Hue bridge does.  I'm not aware of something similar for the Trådfri remote.

> If you reset the remote and one bulb and let them create a new "non trustcenter" network, can that remote then be used in the last step?

I haven't tried that, but that would make the Hue bridge join the ad-hoc network created by the remote, not the network created by deCONZ (or rather by the RaspBee).

wvuyk, 2017-11-18:
Done that tonight also Erik. Thanks for the hint of resetting lights with the dimmerswitch in another thread... saved me lots of time!

mtornblad, 2017-12-01:
I've been playing a bit more with this and extended rest_touchlink.cpp to be able to transmit a "join end device" packet. The idea behind this is to see if an Ikea remote that has been touchlinked works differently than if it has been joined with "classic commissioning". Missing part before I can test it is to calculate and encrypt network key. 

Is there any functionality in the deconz API I can use to do AES encrypt, or do I need to bring in some external library or code for that? According to ZLL spec transaction identifiers from request and response should be merged and used for AES encryption with the "secret" master key!

manup, 2017-12-01:
The problem is that per default deCONZ is running as trust center with default trust center network key. As per ZigBee specification it is not allowed to join devices via touchlink to a trust center network.

The hue bridge would just ignore the command because of that, regardless if it is send by deCONZ or a remote in the deCONZ network.

Therefore deCONZ must be configured to run in `distributed trust center` network mode (trust center address 0xffffffffffffffff). I'm afraid that the whole setup must be done from ground up after that, since it's not possible to transform a trust center network into a distributed trust center network. 

>Is there any functionality in the deconz API I can use to do AES encrypt, or do I need to bring in some external library or code for that? According to ZLL spec transaction identifiers from request and response should be merged and used for AES encryption with the "secret" master key!

Since the network key must be encrypted with the master key we can do that in the RaspBee/ConBee firmware and send it to deCONZ so that the key is available to *rest_touchlink.cpp*, this would have the advantage to use the AES unit on the rf module and it works on every platform without external libraries.

mysteryan, 2017-12-05:
Just a brainstormed thought.  Instead of having Hue bridge join the same network as Deconz, perhaps its possible to migrate the same network into Hue.  Hue has a migration process that allows you to migrate the current zigbee network from their v1 bridge to their v2+ bridge.  I'm not sure if the data is migrated over via zigbee or tcp, but when I performed this, the network was cloned over and v2 bridge had control all without doing anything to the bulbs.  If this process is reproduceable, it might acquire the same network and link keys as the existing deconz network

mtornblad, 2017-12-16:
Uploaded my tests here, https://github.com/mtornblad/deconz-rest-plugin.  After checkout you need to change rest_touchlink.cpp
line 689: I did not get apsCtrl->getParameter(deCONZ::ParamExtendedPANID); to work so this is hardcoded
line 704: ZLL masterkey, google "#DIY lover #ZigBee #Philips #Hue Please RT @travisgoodspeed @stevewoz"

It works to lightlink a Hue hub
Ikea remote and gateway does not work. The only difffence I can see is in some header in the sent package compared to what an Ikea remote sends. if I remember correct the difference I could find was in MAC header -> Frame control -> Frame Type.

First do a scan to get ID of device to link, then use command /api/<apikey>/touchlink/<id>/join

mtornblad, 2017-12-16:
@mysteryan: Could not find any clues in how to use backup file. Some info about how to create and restore backup you have below:

OPTIONS http://192.168.1.159/backup 
http://192.168.1.159/backup
backup options
HEADERS

Content-Type
application/json
Access-Control-Request-Headers
hue-whitelist
BODY

{"devicetype" : "mattias"}

Sample Request
http://192.168.1.159/backup
curl --request OPTIONS \
  --url http://192.168.1.159/backup \
  --header 'access-control-request-headers: hue-whitelist' \
  --data '{"devicetype" : "mattias"}'
PUT http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config 
http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config
create backup
HEADERS

Content-Type
application/json
Access-Control-Request-Headers
hue-whitelist
hue-whitelist
cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4
BODY

{"backup":{"status":"startmigration"}}

Sample Request
http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config
curl --request PUT \
  --url http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config \
  --data '{"backup":{"status":"startmigration"}}'
GET http://192.168.1.159/backup 
http://192.168.1.159/backup
backup
HEADERS

Content-Type
application/json
Access-Control-Request-Headers
hue-whitelist
hue-whitelist
cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4

Sample Request
http://192.168.1.159/backup
curl --request GET \
  --url http://192.168.1.159/backup \
  --header 'hue-whitelist: cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4'
GET http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config 
http://192.168.1.159/api/cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4/config

HEADERS

Content-Type
application/json
Access-Control-Request-Headers
hue-whitelist
hue-whitelist
cBZUo0a9kk0AnhMx6eOyRuGRBcp0CG6XkFaTvTO4

And to restore it is similar to backup but upload to /restore instead of download from /backup

ebaauw, 2017-12-16:
So after the lightlink of the Hue bridge, both the bridge and deCONZ are on the same ZigBee network (PANID and key)?  Does the bridge join the deCONZ network or does deCONZ join the Hue network?  Are both the bridge and RaspBee/ConBee ZigBee gateways (blue nodes in the deCONZ GUI) or did the one become a ZigBee router (yellow node)?

mtornblad, 2017-12-16:
[image]

closed 2017-12-16 by mtornblad

grover, 2018-12-08:
Sorry for bringing this up, but is this a way to get Hue bulbs to update? I.e. can the hue bridge touchlinked to the deCONZ network update the bulbs? Is there any drawback of doing this? Is the bridge and deCONZ still fighting for the same bindings etc.?

manup, 2018-12-09:
Unlike for Ikea due the lack of Philips Hue update files it's needed to update the lights with the Hue bridge.

The bridge can't be linked to the deCONZ network.

>Is the bridge and deCONZ still fighting for the same bindings etc.?

Yes there is little we can do here, since the two controllers can't easily sync settings.