dresden-elektronik/deconz-rest-plugin #267: Requesting sensors over REST results in trailing white space in name

Kane610, 2017-11-12:
Hi!

When requesting sensors over rest I get a trailing white space in the name, or a number at the end is lost. Requesting groups with the same name yield the correct name.

'2': {'modelid': 'TRADFRI wireless dimmer', 'state': {'buttonevent': 3002, 'lastupdated': '2017-11-11T13:38:49'}, 'manufacturername': 'IKEA of Sweden', 'name': 'TRADFRI wireless dimmer ', 'ep': 1,

'lightsequence': [], 'state': {'any_on': False}, 'hidden': False, 'name': 'All lights', 'id': '1', 'lights': ['1', '2'], 'etag': 'e670a437121ca85c9bc3d3b19e6a5e25', 'devicemembership': [], 'class': 'Other', 'action': {'colormode': 'hs', 'xy': [0, 0], 'on': False, 'sat': 127, 'bri': 127, 'hue': 0, 'effect': 'none', 'ct': 0}, 'type': 'LightGroup', 'multideviceids': [], 'scenes': []}, '8443': {'lightsequence': [], 'state': {'any_on': False}, 'hidden': False, 'name': 'TRADFRI wireless dimmer 4'

ebaauw, 2017-11-18:
Probably some glitch when pairing the dimmer.  You can update the name from the REST API, PUT `{"name": "New Name"}` to `/api/username/sensors/xx`.

Kane610, 2017-11-23:
It had also happened with a motion detector from Ikea. If it happens more in the future I will create a new issue

closed 2017-11-23 by Kane610