dresden-elektronik/deconz-rest-plugin #268: Round Xiaomi switch (lumi.sensor_switch) not appearing in REST

djwlindenaar, 2017-11-12:
Hi all,
I know this sensor should be supported as discussed, but I can't seem to get this sensor to join properly. I've tried the mentioned 'rain dance' tactics, like keep pressing the button while network is open and retrying to join the sensor a lot of times. 
It looks like the sensor reports the basic cluster on time and it has the model identifier 'lumi.sensor_switch' in there. Therefore I'd expect the REST plugin to pick up this switch and create the sensor for it. 

Anything I need to try and/or debugging I can do?

Best regards
Daniel

ebaauw, 2017-11-12:
I should report Manufacturer 'LUMI' and indeed the model you mentioned.  There should be an OnOff cluster below the Basic cluster.

For more rain dance magic, power off all your lights (ZigBee routers) and keep the sensor close to the gateway while pairing.  This would ensure the sensor connects directly to the gateway, instead through the mesh network.

micha91, 2017-11-13:
I also have the problem that my new Xiaomi Switches do not appear in the REST-Interface. I am using deCONZ 2.04.86 on Windows and have already one of the Xiaomi Buttons working and in the REST-API as sensor. 
The only differences to the working one are beside the missing sensors in the REST-API that they do not appear as lumi.sensor but with the NWK as name and it is not possible to read the simple descriptor. The clusters of the new sensors are the same as of the old one. Is there something like a whitelist for MAC-Adresses which may needs to be extended?

djwlindenaar, 2017-11-18:
OK, I've found the missing part in my rain dance choreography. After the initial reset through the very small button on the back of the sensor, the one you need a needle for, I now kept pushing on that a couple of times. That made it appear first time. I guess I misunderstood earlier advice to press the button multiple times as the button itself. It should be the small button on the back. 

Now it works fine. Thanks!

closed 2017-11-18 by djwlindenaar