dresden-elektronik/deconz-rest-plugin #270: Hue Firmware Update

Krocko, 2017-11-14:
Today Philips published firmware updates for various lights. Any chance to update the lights with deCONZ?

ebaauw, 2017-11-18:
> Today Philips published firmware updates for various lights

Which lights?  Which versions?  I see no mention of this on the [release notes](http://www2.meethue.com/en-us/support/release-notes/lamps-luminaires), nor in the Hue developer forum.

snozzlebert, 2017-11-18:
On the Dutch website: http://www2.meethue.com/nl-nl/ondersteuning/versie-informatie/lamps-luminaires
I've upgraded my LCT015, LTW012, LWB010 yesterday. All to version 1.29.0_r21169
I think
> Security improvement: proximity based linking with a Hue dimmer switch is now only possible for 30 minutes after the light is powered on.

also affects the Touchlink function in deCONZ.
Possibly it contains changes for Hue Entertainment? (#167)

Krocko, 2017-11-18:
@ebaauw i have read about it in a German forum. 

http://forum.smartapfel.de/forum/thread/675-heute-großes-hue-lampenupdate/

manup, 2017-11-18:
Currently it is not possible to update Hue lights with deCONZ due the lack of official available firmware files.

The changelog suggests that there are no functional improvements, but the Touchlink restriction for 30 min is quite useful and prevents a evil neighbour from stealing your lights :)

ebaauw, 2017-11-18:
Cool, thanks!

> Any chance to update the lights with deCONZ?

Yes, please.  Really not looking forward at resetting all of my 29 Hue lights, connecting them to the Hue bridge for updating the firmware, resetting them, connecting them back to deCONZ, hoping the groups and scenes will be restored correctly.  I'm really tempted to try if the Hue bridge will update the light firmware when it's joined to the deCONZ network (see #266).

I like the security improvement, but it really shows Philips expect the lights to be powered all the time.  In other words: they don't plan on supporting `state.reachable` for traditional wall switches, nor providing sensible power-on defaults.  I wonder if the 30min limitation also holds for the reset through the Hue dimmer switch (holding On and Off buttons for 10 seconds)?

> Possibly it contains changes for Hue Entertainment?

Would they report that as _Reachability improvements of the lights._?  I would expect a bridge firmware update and app update for that as well.

snozzlebert, 2017-12-11:
>> Possibly it contains changes for Hue Entertainment?

>Would they report that as Reachability improvements of the lights.? I would expect a bridge firmware update and app update for that as well.

I'm pretty sure the firmware is related to Hue Entertainment.
There is a difference in ```capabilities``` in the bridge response before and after the firmware update of a bulb.
Before:
```
{
    "state": {
        "on": true,
        "bri": 254,
        "hue": 8418,
        "sat": 140,
        "effect": "none",
        "xy": [
            0.4573,
            0.41
        ],
        "ct": 366,
        "alert": "none",
        "colormode": "ct",
        "mode": "homeautomation",
        "reachable": true
    },
    "swupdate": {
        "state": "noupdates",
        "lastinstall": null
    },
    "type": "Extended color light",
    "name": "Hue color candle 1",
    "modelid": "LCT012",
    "manufacturername": "Philips",
    "capabilities": {
        "streaming": {
            "renderer": false,
            "proxy": false
        }
    },
    "uniqueid": "00:17:88:01:03:00:0a:0a-0b",
    "swversion": "1.15.2_r19181",
    "swconfigid": "79A02AB9",
    "productid": "Philips-LCT012-1-E14ECLv1"
}
```

After:

```
{
    "state": {
        "on": false,
        "bri": 254,
        "hue": 0,
        "sat": 0,
        "effect": "none",
        "xy": [
            0.3804,
            0.3768
        ],
        "ct": 366,
        "alert": "none",
        "colormode": "ct",
        "mode": "homeautomation",
        "reachable": true
    },
    "swupdate": {
        "state": "noupdates",
        "lastinstall": "2017-12-06T20:42:04"
    },
    "type": "Extended color light",
    "name": "Hue color candle 1",
    "modelid": "LCT012",
    "manufacturername": "Philips",
    "capabilities": {
        "streaming": {
            "renderer": true,
            "proxy": true
        }
    },
    "uniqueid": "00:17:88:01:03:00:0a:0a-0b",
    "swversion": "1.29.0_r21169",
    "swconfigid": "EA9279F9",
    "productid": "Philips-LCT012-1-E14ECLv1"
}
```

```streaming.renderer``` and ```streaming.proxy``` are ```true``` after updating

ebaauw, 2017-12-11:
Yes.  What Hue bridge firmware are you on?  See also https://github.com/ebaauw/homebridge-hue/issues/217#issuecomment-345854006.

snozzlebert, 2017-12-11:
Version is 1.22.0

ebaauw, 2017-12-11:
Ah, I see they published it Nov 27.

mathos77, 2018-04-19:
Not that I have any issues with my current Hue bulbs, but it would be nice if we would somehow be enabled to update the lights via deCONZ.

If we would know where the firmware files are, would we then be able to update the bulbs ?

ebaauw, 2018-04-19:
That depends on whether they’re encrypted.

timbru31, 2018-11-20:
If it helps I could do some network sniffing and provide the request and responses from the Hue App / Bridge, I have to update some hue bulbs and can borrow a bridge.

manup, 2018-11-20:
Would be interesting to access the files not sure if the url can be sniffed when https is used and it might be within the content body.

Hypfer, 2019-03-20:
This bot does more harm than good tbh

timbru31, 2019-04-13:
Sorry for the silence - I had no luck back in November (had to bring the bridge back the day after) but purchased a used Hue bridge for small money, so I have more time to sniff the network traffic now :)

timbru31, 2019-04-14:
It seems that the Hue App is actually downloading the firmware, at least this is the only interesting package I've found when my Hue app was searching for new software updates:
[image]

ebaauw, 2019-04-14:
It's the Hue bridge that downloads the firmware, not the app.  The app just issues a REST API call to the bridge to start the search.

timbru31, 2019-04-14:
Yeah captured that now, too: `{"swupdate":{"checkforupdate":true}}` is the payoad in the PUT request.

timbru31, 2019-04-15:
My bridge made a request to http://fds.dc1.philips.com/firmware/ZGB_100B_010C/16783872/100B-010C-01001A00-ConfLight-Lamps_0012.zigbee (267kb)

- No auth needed. Any idea what this file could be? `strings` and a HexEditor did not yield any useful information.

ebaauw, 2019-04-15:
Judging by the filename and size, that's a firmware image alright.
```
100B-010C-01001A00-ConfLight-Lamps_0012.zigbee
|    |    + image file version
|    +- image type (device type)
+- manufacturer code 0x100b = Philips
```
These match the fields in the _STD OTAU Plugin_ panel of the deCONZ GUI, see e.g. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96#issuecomment-321022808.

The image type and file version match the current (1.46.13_r26312) firmware for the Color Temperature bulbs and spots (LTWxxx).

Now the $100M question is whether this file is encrypted.  If it isn't, you should be able to download it, feed it do the OTAU plugin and update the firmware from deCONZ.  If it is, the file is useless without the decryption key.

timbru31, 2019-04-15:
I have sniffed the NWK of the Hue bridge so I can capture the Zigbee packages, too. And the ZLL OTA cluster is (AFAIK) without encrypted firmware?

manup, 2019-04-15:
**Awesome!**

Yes this is a valid OTA image and encrypted but that's ok. OTA files should be encrypted since they are transferred transparently.

[image]

manup, 2019-04-15:
Note decryption usually is done on the receiving device.

Some vendors don't encrypt their files though (not saying any names here..).

ebaauw, 2019-04-15:
> Note decryption usually is done on the receiving device.

Cool, didn't know that.

The URI seems to redirect to the AWS cloud.  They don't support browsing the directory structure (if the URI path even reflects one), so we might have to re-sniff the Hue bridge for every type and version.  Unless there's like an index file containing links to the firmware files, like IKEA has.

I suppose the Hue bridge only requests the firmware for connected lights?

manup, 2019-04-15:
>I suppose the Hue bridge only requests the firmware for connected lights?

Likely, maybe something like *pihole* in between can help to figure out more URLs.

timbru31, 2019-04-15:
> The URI seems to redirect to the AWS cloud. They don't support browsing the directory structure (if the URI path even reflects one), so we might have to re-sniff the Hue bridge for every type and version. Unless there's like an index file containing links to the firmware files, like IKEA has.

The only URL that's not returning a 404 is http://fds.dc1.philips.com/ with a (*strange?*) response of `ecdprd`  
Is it actually an S3 bucket? There are some tools to scan them :)

> I suppose the Hue bridge only requests the firmware for connected lights?

Seems to be true. On the other hand I've only got the E14 lights.

> Likely, maybe something like pihole in between can help to figure out more URLs.

I've got one running, too. I can take some deeper looks into the logged queries.

timbru31, 2019-04-15:
> The image type and file version match the current (1.46.13_r26312) firmware for the Color Temperature bulbs and spots (LTWxxx).
Firmware matches my updated LCT012, too

timbru31, 2019-04-15:
Other noteworthy philips domains (sadly pihole logs only the domain names not paths/queries since it's only my DNS resolver)
- dcp.dc1.philips.com
- www.crsc.philips.com
- www.ecdinterface.philips.com
- www.crsc.lighting.philips.com
- philipslighting.d3.sc.omtrdc.net
- philipslighting.tt.omtrdc.net
- diagnostics.meethue.com

ebaauw, 2019-04-15:
LCT012 is a gamut-C extended colour light, which, I understand (I don't have any of these myself) has 5-channels: RGB+CW/WW.  So it would make sense to use the same firmware as for the colour temperature lights (with two channels: CW/WW), where the RGB part only used for lights that support this.

From what I've seen, the gamut-B extended colour lights (RGB but different colours, optimised for colour temperature), the gamut-A colour lights (also RGB, with other different colours, optimised for colours), and the on/off lights all have different firmware builds.

timbru31, 2019-04-15:
Back to the network traffic:
after resetting the bridge to factory state and after the PUT to check for updates it always connects to:
- `http://www.ecdinterface.philips.com:80/DevicePortalICPRequestHandler/RequestHandler.ashx`  

After factory reset it also connects to:
- `http://www.ecdinterface.philips.com:80/NotificationService/RequestHandler.ashx`
- `http://dcp.dc1.philips.com:80/DcpRequestHandler/index.ashx`

timbru31, 2019-04-15:
Given that Philips encrypted the OTA files for older firmware, I think that one I sniffed is encrypted too :/
See https://www.blackhat.com/docs/us-16/materials/us-16-OFlynn-A-Lightbulb-Worm.pdf (start at page 24 and 64)

timbru31, 2019-04-16:
Would a sniffed OTA upgrade (Zigbee/IEEE 802.15.4 traffic) help any further? I've captured the OTA process - if possible I could send the pcap file and NWK key. @manup @ebaauw

manup, 2019-04-16:
I think on Zigbee level all should be fine as the data is transferred transparently, only need more URLs for various firmware files :)

manup, 2019-04-19:
@timbru31 how do you sniff the traffic of the Hue App? Tried some apps like "Packet Capture", but couldn't figure out the snooping in ssl connections.

ebaauw, 2019-04-19:
To find the firmware location, you need to sniff the (apparently unencrypted) traffic from the Hue bridge to the cloud, not the (https) traffic from the Hue app to the Hue bridge.

The old (round) bridge doesn’t support https, so the Hue app can still use plain http.  I suspect/hope that if you block port 443 in your firewall, the app will fallback to plain http for the new (square) bridge as well.  Haven’t tried that myself, though.

timbru31, 2019-04-19:
@manup My sniffing setup is the following:
* Hue bridge (I've got the round v1 one): 
  - Connected to my MacBook with an Ethernet cable
  - Enabled Internet sharing from my WiFi to the Thunderbolt adapter
  - Captured all traffic from bridge100 with Wireshark (the created interface from the Internet Sharing)
  - (You could also set the proxy address with the Hue API to your burp instance, see below)
* Zigbee sniffing:
  - Using a CC2531 USB Stick with https://github.com/andrewdodd/ccsniffpiper
  - Extracted the NWK from the Touchlink pairing
* Hue App sniffing (iOS):
  - Installed Burp Suite (https://portswigger.net/burp/)
  - Set my MacBook as the proxy in the iOS settings
  - Installed the SSL root cert to do a MITM - some apps do SSL cert checking, some don't :) (Spoiler: Hue does not)
  - Have Wireshark running on the WiFi adapter, too, just in case Burp misses something since iOS uses my MacBook as a proxy

---

@ebaauw True with HTTP, but the payload is still AES encrypted and I assume we won't find the key that easily. See attached pcap: [bridge_dump_factory_reset.pcapng.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3098147/bridge_dump_factory_reset.pcapng.zip)

manup, 2019-04-20:
Interesting approach with the Internet sharing, gonna try this next. Looks like only the bridge needs to be sniffed.

I've brought some of my Hue devices to a round v1 Hue bridge, and started update search yesterday. Looked like nothing happened, but after a few hours updates there shown for all devices:

* Motion sensor SML001
* Hue beyond LLM001
* E27 Lux LWB004
* E27 color light LCT001
* GU10 color light  LCT003

When I powered off all devices to not start the update :)

Meanwhile I found another way to sniff the bridge traffic via the Telekom Speedport W 724V router capture feature `http://192.168.2.1/html/capture.html`.

The bridge update state is back to `1` *"Downloading available updates"* and no updates are shown yet. The router sniffer is running, hopefully today brings some new URLs.

Also added the Hue Dimmer switch to the bridge.

timbru31, 2019-05-01:
> Meanwhile I found another way to sniff the bridge traffic via the Telekom Speedport W 724V router capture feature http://192.168.2.1/html/capture.html.

Nice! The FRITZ!Box supports such a feature, too - but it's well hidden :)

---

I've got my hands on a new Hue Bridge v2 (square), on the first connection to the internet it immediately downloaded this file: http://fds.dc1.philips.com/firmware/BSB002/1931140050/BSB002_1931140050.product.RSA_prod_01.fw2, I assume that's the most recent Bridge firmware (not yet with a changelog, the only (German) source regarding this version is: https://hueblog.de/community/frage/gibt-es-infos-zum-update-der-bridge-auf-version-1931140050/)

Cool thing: the naming follows the scheme `http..../BSB002/<fw-version>/BSB002_<fw-version>.product.RSA_prod_01.fw2` (for example an older firmware link: http://fds.dc1.philips.com/firmware/BSB002/1931069120/BSB002_1931069120.product.RSA_prod_01.fw2)

Besides that I've gained root on the box (http://colinoflynn.com/2016/07/getting-root-on-philips-hue-bridge-2-0/, https://blog.andreibanaru.ro/2018/03/27/philips-hue-2-1-enabling-wifi/) and have access to the unencrypted `ipbridge` binary (the one binary that does everything from being a webserver to sending commands to the Zigbee Soc) - now we need to understand how the AES key is constructed to decrypt the connection to www.ecdinterface.philips.com (where I assume the firmware info is stored/returned in the body payload)

ebaauw, 2019-05-01:
> Nice! The FRITZ!Box supports such a feature, too - but it's well hidden

Brilliant!

Markkuuss, 2019-05-13:
@manup: Have you found any new URLs to the following hue devices?
> * Motion sensor SML001
> * Hue beyond LLM001
> * E27 Lux LWB004
> * E27 color light LCT001
> * GU10 color light  LCT003

ebaauw, 2019-06-03:
And for the Hue motion sensor?  See #1559.

robertlandes, 2019-06-03:
Btw. I can confirm, that zigbee firmware file `100B-010C-01001A00-ConfLight-Lamps_0012.zigbee`  from above updated the following lamps to latest firmware version through OTA:

- White & Color Ambience E27 - Model LCT015 > Version 1.46.13_r26312
**and**
- White E27 - Model **LWB010** > Version 1.46.13_r26312

The strange thing is, that I thought it didn't work properly, because Phoscon / REST API was still showing old firmware version for my LWB010 lamps, which I didn't pair with my Hue Bridge for FW update before pairing it to deCONZ. So last weekend I reset one of the bulbs paired it to my Hue Bridge to do the FW update, just to see, that it already has latest firmware. So repaird it back to deCONZ (without deleting it) and it still showed old FW version. After deleting it and pairing it again, it showed up as new light with the correct firmware version. So it seems, that there is an Issue with updating FW versions after OTA update. 

Probably will need to open a separate issue for that.

manup, 2019-06-03:
>@manup: Have you found any new URLs to the following hue devices?

Sadly not yet, my last attempt a while ago with the Hue bridge v1, over a few days, didn't yield any new URLs, probably my Hue devices  are too old or too new to get updated.

>Btw. I can confirm, that zigbee firmware file 100B-010C-01001A00-ConfLight-Lamps_0012.zigbee from above updated the following lamps to latest firmware version through OTA:

Congrats this is the first time a Hue device was updated via deCONZ :)

> So it seems, that there is an Issue with updating FW versions after OTA update.

Once the firmware version was queried it isn't refreshed for Hue lights. We can active this in future releases, it's already active for certain other devices which could be updated. 

To manually query the firmware version you can read the Basic cluster attributes in the deCONZ GUI.

ebaauw, 2019-06-03:
> After deleting it and pairing it again, it showed up as new light with the correct firmware version. So it seems, that there is an Issue with updating FW versions after OTA update.

You need to force-read the _Basic_ cluster attributes before the GUI and REST API show the new firmware (and date code?).  If the new firmware exposes a different profile or device id (e.g. upgrade from ZLL to Zigbee 3.0, which uses ZHA), you need to delete and re-pair the device.  Just force-reading the _Simple Descriptors_ isn't enough to update the GUI.

l-mb, 2019-06-21:
Hey, I wonder if someone has actually tried reaching out to Philips. Sorry if this is a dumb question, but have they officially denied publishing their firmware images more readily? I need to update my bulbs and this does seem like a project I could try to play with next week, but first I'd like to know whether the easy path has been tried ;-)

timbru31, 2019-06-27:
Personally I haven't asked Philips 🙈

rb2k, 2019-07-04:
@timbru31 any chance you ever tried to use DNS poisoning to redirect the hue requests to a local MITM proxy?

robertlandes, 2019-07-10:
> @timbru31 any chance you ever tried to use DNS poisoning to redirect the hue requests to a local MITM proxy?

Yes, that would be interesting. I think https://mitmproxy.org/ can probably be a tool to achieve this. I would test it, I just don’t have any Hue hardware which is not the current firmware version...

rb2k, 2019-07-10:
That or https://www.charlesproxy.com/ on macOS.
I guess we'd have to hope that they don't do certificate pinning in the app / firmware.

I sadly don't have a bridge anymore so I'm afraid I am not very useful :)

timbru31, 2019-07-11:
I’ve already MITM myself with Burp. The problem is that the payload of the https request is AES encrypted, so we need to find the key in the bridge/or the bridge firmware.

So far I’ve had no time to look into this.

rb2k, 2019-07-11:
Just to clarify do you mean that:

a) the MITM worked and you can see the HTTP interaction, but the body of the HTTP response is a binary blob.

or:

b) Do you mean that the whole communication was SSL encrypted because it's HTTPS

For b) charlesproxy installs an ssl certificate on the phone and will use it to re-encrypt the traffic
For a) from what I understand, it's fine to have the files themselves be encrypted since the bulbs will handle the decryption

timbru31, 2019-07-11:
a) MITM, including SSL works, but besides the HTTPS the actual payload is AES encrypted.

>For a) from what I understand, it's fine to have the files themselves be encrypted since the bulbs will handle the decryption

The OTA files are not the problem. But the AES encrypted request to ecdinterface.philips.com is the problem. Probably the response contains the URLs of the firmwares (they are located at http://fds.dc1.philips.com/firmware/)

izzy84075, 2019-07-17:
Does anybody know what the latest version for LWB004 lights is? I gave away my Hue hub a while ago, so I have no way of sniffing out it's secrets.

ooii, 2019-09-04:
Hi, long time since I did not visit deconz repo.
I read through this thread and it seems Hue bulbs can be updated from Deconz, cf. `Congrats this is the first time a Hue device was updated via deCONZ :)`.
Is there any procedure I can follow to do that? I have maybe 40 bulbs and don't want to unpair/pair/unpair/pair them. Thanks.

MattL0, 2019-09-22:
+1 i am currently updating a lot of my hue device... this is painful haha.... ( but can study at the same time). The task is almost finished now ( why the he** does it take 45  in per device???)

any device are available for direct fw update via deconz?

247GradLabs, 2019-10-04:
@manup Are there any updates regarding this feature? I would love to connect all my devices to deCONZ, but having no updates for all my Hue bulbs is no option for me at the moment.

flobernd, 2019-10-04:
@manup Are there any updates regarding this feature? I would love to connect all my devices to deCONZ, but having no updates for all my Hue bulbs is no option for me at the moment.

qvicksilver, 2019-10-19:
Tried reaching out on the Facebook page:
https://www.facebook.com/HuePhilips/posts/1910598109072064?comment_id=1910732495725292&reply_comment_id=1910939499037925&notif_id=1571500554024772&notif_t=feed_comment

Followed up on that with a post on the forum:
https://developers.meethue.com/forum/t/firmware-downloads/6242

Feel free to join in and create some buzz.

ottelo9, 2020-01-12:
> Judging by the filename and size, that's a firmware image alright.
> 
> ```
> 100B-010C-01001A00-ConfLight-Lamps_0012.zigbee
> |    |    + image file version
> |    +- image type (device type)
> +- manufacturer code 0x100b = Philips
> ```
> 
> These match the fields in the _STD OTAU Plugin_ panel of the deCONZ GUI, see e.g. [#96 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96#issuecomment-321022808).
> 
> The image type and file version match the current (1.46.13_r26312) firmware for the Color Temperature bulbs and spots (LTWxxx).
> 
> Now the $100M question is whether this file is encrypted. If it isn't, you should be able to download it, feed it do the OTAU plugin and update the firmware from deCONZ. If it is, the file is useless without the decryption key.

How can I convert the Hex 0x01001a00 file version to the Version String 1.46.13_r26312?

mories76, 2020-02-12:
I've bought myself a Hue bridge, ripped it open, became root. 
Connected both my iPhone and the HUE via InternetSharing on my iMac

This way I could capture an url for the latest HUE motion sensor firmware.
http://fds.dc1.philips.com/firmware/ZGB_100B_010D/1107323831/Sensor-ATmega_6.1.1.27575_0012.sbl-ota (I will try to update one of my Motion sensor with this firmware via deconz)
While the bridge is downloading/updating the firmware file it is stored as /tmp/image.fw

I've also found an url for the Bridge firmware. I think it's encrypted, not sure yet. There is a pem file somewhere on the filesystem, which might be used for decrypting the image.

There doesn't seem to be an easy way to a json like the way Ikea does.
Still poking around.

[134 more replies]