dresden-elektronik/deconz-rest-plugin #271: IKEA color bulb returning invalid values in REST API

wvuyk, 2017-11-16:
Hi,

Am on version 2.04.89 here. When I look at the returned values for the IKEA color bulb, it is suggested the colormode="ct" where no ct vlaues are returned - as far I understand this light is not enabled for ct at all even, only xy? In the returned values, the xy is returned correctly...

`{
    "ctmax": 454,
    "ctmin": 250,
    "etag": "b717aaeed4f8c53acc9b917f9adfa461",
    "hascolor": true,
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 CWS opal 600lm",
    "name": "Tradfri Color",
    "state": {
        "alert": "none",
        "bri": 183,
        "colormode": "ct",
        "effect": "none",
        "hue": 0,
        "on": true,
        "reachable": false,
        "sat": 0,
        "xy": [
            0.459796,
            0.411618
        ]
    },
    "swversion": "1.3.002",
    "type": "Color light",
    "uniqueid": "00:0b:57:ff:fe:a7:e9:46-01"
}`

ebaauw, 2017-11-16:
Yes, see #218, point 4.

wvuyk, 2017-11-16:
Erik, sorry for that, have been searching if this was known, but did not find it..

Will close this one as it is a double now

Wim

closed 2017-11-16 by wvuyk