dresden-elektronik/deconz-rest-plugin #272: Group events on web socket doesn't contain much information

Kane610, 2017-11-17:
Hi!

When receiving events about groups over web socket there is no updated state information except whether it is on or off.

I'd like to see updated brightness and color and all the other aspects that might've changed.

ebaauw, 2017-11-18:
> When receiving events about groups over web socket there is no updated state information except whether it is on or off.

That's because that's the only state information available for groups.  Admittedly, this _is_ confusing, imho due to poor design (initially by Philips) of the REST API.

In ZigBee there's different messages for sending a command to a device, e.g. _On_, _Off_,  _Toggle_, _On for n seconds_ versus querying the device attributes, e.g. _OnOff_ which indicates whether the light is currently on of off.  You could say a command is write-only, whereas an attribute is read-only (although some attributes used for configuration are read/write).

The REST API provides a `state` resource per light, with read-write attributes handling both (write-only) commands as well as (read-only) attribute reporting.  PUTting `{"on": true}` to a light state actually sends the _On_ command.  When polling the light, the gateway queries the _OnOff_ attribute and updates the `on` attribute of the `state` resource accordingly.

There are commands without a corresponding attribute, e.g. the _Identify_ command (which makes the light blink) is mapped to the `state` resource `alert` attribute in the REST API.  Hence `state.alert` reports the last value PUT, instead of the actual light state.  Imho, a more consistent REST API design would have offered a (PUT-only) `/lights/1/commands/identify` resource, instead of an `alert` attribute of the `state` resource.  This would offer (PUT-only) resources like `commands/on`, `commands/off` and `commands/toggle` and the `state` resource would become GET-only.  Similarly, you would get `commands/bri` and `commands/bri_inc` resources and only a `bri` attribute to the `state` resource, instead of a (PUT-only!) `bri_inc` attribute.

In ZigBee, a group is just a multicast address, an efficient way to send a command to multiple lights. There's no group device, and no group attributes.  Consequently, a group has an `action` resource instead of a `state` resource.  However, instead of modelling the commands (e.g. a PUT of `/groups/1/commands/on`), they modelled it like attributes (e.g. PUT '{"on":true}' to `/groups/1/action`).  Like the `alert` attribute of a light `state` resource, the attributes of a group `action` only report the last values PUT.

The API also supports a `state` resource for groups.  The gateway (or Hue bridge) computes the attributes for this resource by aggregating the corresponding attribute from the state resources from the member lights.  Currently this is only done for the light `state.on`, resulting in a group `state.any_on` and `state.all_on`.  This is what's being reported to the web socket.

> I'd like to see updated brightness and color and all the other aspects that might've changed.

Web socket notifications for `group.action` would only indicate that some other app has sent a group command; they would not reflect any changed light states.

It's hard to define what a group `state.bri` would actually mean.  The average brightness of the member lights?  What about on/off lights that don't have a brightness?  Should a 1000lm light count the same as a 400lm light?  An average colour or average colour temperature would even be more complicated.

Kane610, 2017-11-21:
I understand, I've done a work around that works well enough. Thanks for your answer!

closed 2017-11-21 by Kane610