dresden-elektronik/deconz-rest-plugin #274: Option to disable deconz ssdp server

mariusmotea, 2017-11-17:
Hi,

So far i was not bother so much by the ssdp conflict with my hue emulator, but currently i'm using a kody cinema plugin that perform every time ssdp detection and it fail. The procedure is complicated, i need to stop deconz, disable deconz support in my emulator config (otherwise will fail to start because deconz is not reachable) and then restart the hue emulator. After successful detection i need to undo all these changes and this is very frustrating for me.

Thanks.

manup, 2017-11-19:
With the last commit you can start deCONZ with `--upnp=0` to disable the SSDP discovery.

mariusmotea, 2017-11-19:
Thanks, what will be the release that will contain this commit?

manup, 2017-11-19:
That will be 2.04.90 :)

mariusmotea, 2017-11-20:
Works, thanks!

closed 2017-11-20 by mariusmotea