dresden-elektronik/deconz-rest-plugin #276: Beta App: advanced rules?

markuspf, 2017-11-17:
I like how I can easily configure a motion sensor (philips hue) assigned to a group.

But, in the HUE app I can configure a light level when motion detection triggers, and different behaviours for different times of day. Is this a planned feature?

In general I am a bit confused by this "rules" business: *Where* are they executed? Is this something that deCONZ does? Is there some rules that are actually programmed into the Zigbee devices (what kinds of restrictions does that bring)? Is there any documentation for rules?

manup, 2017-11-18:
> But, in the HUE app I can configure a light level when motion detection triggers, and different behaviours for different times of day. Is this a planned feature?

Absolutely, the time range and also light level, or even easier `dark` attribute of Philips and IKEA sensors will be available for motion sensor controls.

There will be an update in the next days which fixes some bugs in the sensor control settings together with an early `dark`, support.

Setting time ranges will come later but already have a high priority.

> In general I am a bit confused by this "rules" business: Where are they executed? Is this something that deCONZ does? Is there some rules that are actually programmed into the Zigbee devices (what kinds of restrictions does that bring)? Is there any documentation for rules?

All rules are executed by deCONZ which allows broad range of applications, the ZigBee devices wouldn't be able to handle all this due the low resources.

Current documentation is here: http://dresden-elektronik.github.io/deconz-rest-doc/

It is quite incomplete and will get more comprehensive in the next months.

ebaauw, 2017-11-18:
> >In general I am a bit confused by this "rules" business: Where are they executed? Is this something that deCONZ does? Is there some rules that are actually programmed into the Zigbee devices (what kinds of restrictions does that bring)? Is there any documentation for rules?

> All rules are executed by deCONZ which allows broad range of applications, the ZigBee devices wouldn't be able to handle all this due the low resources.

Typically, sensors (and switches) work in two different ways, reacting to changes:
1. They report their new state (contact open, button pressed, motion detected, lightlevel measured, ...); and/or
2. They issue a command (turn on, increase brightness, recall scene, ...).

In 1), the state is reported to the gateway (or Hue bridge), where typically it’s used to trigger rules.  In 2), the command is sent to a group, and the lights in that group receive the command without involvement of the gateway.  The deCONZ gateway eavesdrops on these group commands, reverse engineers them into state changes, and updates the sensor resource, so they can be used to trigger gateway rules as well.

Which way is used depends on the device: e.g. Philips uses 1), but it also supports 2) when pairing the dimmer switch directly to a light; IKEA uses 2).  So, the Hue motion sensor reports like: motion detected, motion no longer detected, current lightlevel, current temperature.  The duration attribute determines when motion no longer detected is reported.  The Trådfri motion sensor sends a command: turn lights on for x minutes, turn lights on for x minutes, but only when they’re already on.  The dials on the sensor set the x, and the “only when already on” part, which deCONZ reverse engineers into `duration` and `dark`.

Each way has its advantages: way 1 is more powerful and provides more fine-grained info to the gateway.  That’s why the Hue dimmer switch supports different `buttonevent` values for press, hold, release, long release.  Way 2 works without a gateway, or when the gateway is down.

rtenklooster, 2018-01-21:
It would be great if open/ close sensors could be used in the automations. 
I don't know about other then Xiaomi door sensors, but I am unable to use the Xiami / Aqara sensors as triggers for automations. Would be great if it would be possible to use them.

ebaauw, 2018-01-21:
I have gateway rules firing by the Aqara door sensors.  I also get HomeKit notifications and automations when exposing the Aqara through homebridge-hue.  Just use:
```json
  "conditions": [
    {
      "address": "/sensors/xx/state/open",
      "operator": "eq",
      "value": "true"
    },
    {
      "address": "/sensors/xx/state/open",
      "operator": "dx"
    }
  ],
```
Note that `"value"` takes a string value - not a boolean.

closed 2019-03-30 by stale[bot]