dresden-elektronik/deconz-rest-plugin #277: Project ERROR: Unknown module(s) in QT: serialport

philko123, 2017-11-19:
I try to install the plugin on a raspi3 with qt5. i use a conbee on usb0

--

No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie

--

libqt5clucene5/oldstable,now 5.3.2-3 armhf [installed,automatic]
libqt5concurrent5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5core5a/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5dbus5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5designer5/oldstable,now 5.3.2-3 armhf [installed,automatic]
libqt5designercomponents5/oldstable,now 5.3.2-3 armhf [installed,automatic]
libqt5gui5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5help5/oldstable,now 5.3.2-3 armhf [installed,automatic]
libqt5network5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5opengl5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5opengl5-dev/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5printsupport5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5qml5/oldstable,now 5.3.2-4+rpi1+b1 armhf [installed,automatic]
libqt5quick5/oldstable,now 5.3.2-4+rpi1+b1 armhf [installed,automatic]
libqt5quickwidgets5/oldstable,now 5.3.2-4+rpi1+b1 armhf [installed,automatic]
libqt5serialport5/oldstable,now 5.3.2-2 armhf [installed,automatic]
libqt5sql5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5sql5-sqlite/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5svg5/oldstable,now 5.3.2-2 armhf [installed,automatic]
libqt5test5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5webkit5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5websockets5/oldstable,now 5.3.2-5 armhf [installed,automatic]
libqt5websockets5-dev/oldstable,now 5.3.2-5 armhf [installed]
libqt5widgets5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
libqt5xml5/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]
qt5-default/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed]
qt5-qmake/oldstable,now 5.3.2+dfsg-4+rpi1 armhf [installed,automatic]

philko123, 2017-11-20:
`sudo apt-get install libqt5serialport5-dev`

closed 2017-11-20 by philko123

manup, 2017-11-20:
Hi, thanks for figuring this our, will add the dependency in the dev debian package so a more helpful warning will be displayed after installing.

bj97301, 2020-03-04:
How about for a mac?