dresden-elektronik/deconz-rest-plugin #278: Xiaomi "Original" motion sensors

ensamresande, 2017-11-20:
I have picked up some of the older/"Original"/Mijia Xiaomi motion sensors but am a bit lost with how I get them to appear in the REST API. I think I have managed to pair them but this is as far as I can get. Am I missing something or are they not yet supported?

[image]
[image]

ensamresande, 2017-11-21:
Also forgot to mention that I'm using 2.04.86 (latest Windows release?)...

ensamresande, 2017-11-21:
I also have the [older oval Xiaomi door sensor](https://www.gearbest.com/smart-light-bulb/pp_257677.html) as well as the [older round Xiaomi temperature/humidity sensor](https://www.fasttech.com/products/1450/10021232/4586300)). Both of these join my network (as well as a Hue light):

[image]

But I can only see the temperature/humidity sensor in the REST interface:

```
  "sensors": {
    "1": {
      "config": {
        "on": true,
        "reachable": true
      },
      "ep": 1,
      "etag": "9beaa5323d7a07c0da5665cf2d86c21b",
      "manufacturername": "LUMI",
      "modelid": "lumi.sensor_ht",
      "name": "Temperature 1",
      "state": {
        "lastupdated": "2017-11-21T20:06:59",
        "temperature": 3078
      },
      "type": "ZHATemperature",
      "uniqueid": "00:15:8d:00:01:dc:c4:f1-01-0402"
    },
    "2": {
      "config": {
        "on": true,
        "reachable": true
      },
      "ep": 1,
      "etag": "9beaa5323d7a07c0da5665cf2d86c21b",
      "manufacturername": "LUMI",
      "modelid": "lumi.sensor_ht",
      "name": "Humidity 2",
      "state": {
        "humidity": 7064,
        "lastupdated": "2017-11-21T20:06:59"
      },
      "type": "ZHAHumidity",
      "uniqueid": "00:15:8d:00:01:dc:c4:f1-01-0405"
    }
  }
```

Is there a breaking difference between these older "Mijia" sensors and and the newer "Aqara" sensors mentioned in e.g. #281?

manup, 2017-11-21:
From the screenshot they look very similar and should also be added to the REST API since the model id starts with `lumi.sensor_motion` too, it is very important to add them *while the network is open* (sensor search is active).

The Aquara motion sensors do have a small button which can be pressed like 5 seconds and the sensor will blink and join the network. Then the button should be pressed again and the sensor will be added.

The windows version 2.04.90 is also up now:
https://www.dresden-elektronik.de/deconz/win/

ensamresande, 2017-11-21:
Just to clarify, _while the network is open_ refers to the Edit -> Permit Join in the Windows version (or the "Open Network" in the web interface)? I just tried is the following:

1. Fresh install of 2.04.90
2. Network is currently empty
3. Set network to open.
4. Hold the button on the motion sensor.
5. After ca 3 seconds, it blinks, and I release the reset button
6. I then press the button repeatedly once per second until I see the device appear in the network and the basic and Occupancy sensing cluster appears.
7. Check REST API, no sensors.

If I repeat step 3-6, the device will be replaced with a new one with same MAC and different NWK.

Is there something else I can try? Some debug output I can enable?

I believed the old Mijia were the same as the new Aqara sensors, just lacking the luminance sensor. However the trouble I have getting these to work is making me suspect otherwise. :disappointed:

manup, 2017-11-21:
> Just to clarify, while the network is open refers to the Edit -> Permit Join in the Windows version (or the "Open Network" in the web interface)? I just tried is the following:

Sorry I should have mentioned this, it's the `Open Network` option from the web interface. Opening the network in deCONZ window will let the sensor join but won't add it to the REST API.

ensamresande, 2017-11-22:
Woohoo. :smile: 

```
{
    "1": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "65c6a81b8431f83bfa7f8d1bfe64f890",
        "manufacturername": "LUMI",
        "modelid": "lumi.sensor_motion",
        "name": "Presence 1",
        "state": {
            "lastupdated": "2017-11-22T06:50:51",
            "presence": true
        },
        "type": "ZHAPresence",
        "uniqueid": "00:15:8d:00:01:23:19:aa-01-0406"
    },
    "2": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "cbd76bd6c35cddffa344687887f6aff9",
        "manufacturername": "LUMI",
        "modelid": "lumi.sensor_motion",
        "name": "Presence 2",
        "state": {
            "lastupdated": "2017-11-22T06:51:43",
            "presence": true
        },
        "type": "ZHAPresence",
        "uniqueid": "00:15:8d:00:01:21:c5:b1-01-0406"
    }
}
```

Once I did it via the web interface it worked the first time. :smile:

closed 2017-11-22 by ensamresande