dresden-elektronik/deconz-rest-plugin #279: Missing bridgeid after moving to new Pi

s4y, 2017-11-21:
I recently moved my RaspBee from a first generation Pi to a new Pi 3.

Since the move, the deCONZ has reported its bridgeid as `0000000000000000`. This breaks at least [homebridge-hue](https://github.com/ebaauw/homebridge-hue), so it leads me to think that it's not expected behavior. I just upgraded to 2.04.90, and it seems to be the same. I've also tried resetting the bridge, no change.

What can I try next? Everything else seems to work, except for one Hue A19 that won't pair with the bridge.

ebaauw, 2017-11-21:
The zero bridgeid means deCONZ hasn’t yet read the mac address of the RaspBee or ConBee.  Afaik it should read it after some time automatically.

s4y, 2017-11-21:
Thanks. Yeah, I saw the comment about that in the homebridge-hue code :-). Unfortunately the bridge has been turned on for several days now, with a couple of restarts in between. I think it’s had many chances to read it and failed to do so.

s4y, 2017-12-02:
For posterity, following some steps in #280 fixed this for me. Specifically, I did this (but some of the steps might be unnecessary):

1. Stop headless deCONZ:
    ```
    sudo systemctl stop deconz
    sudo systemctl stop deconz-init
    sudo systemctl stop deconz-update
    ```
2. Run regular deCONZ. In my case I used x11vnc, which crashed frequently, so I did it in a loop and my client auto-reconnected each time it died 🙃:
    ```
    while sleep 1; do sudo x11vnc -create -ncache_cr -passwd foo; done
    ```
3. Open the network settings (button next to "In Network").
4. Check the _Custom MAC Address_ checkbox.
4. Enter the MAC address from the RaspBee's label into the _MAC Address_ field.
5. Copy and paste that address into the _TC Address_ field.
6. Click _Save_ (the _TC Link Key_ field should now have a nonzero value).
7. Close network settings.
8. Click _Leave_.
9. Wait 5 seconds.
10. Click _Join_
11. Open the network settings.
12. *Un*check _Custom MAC Address_
13. Click *Save*
14. Reboot the Pi (`sudo reboot`) to put everything in a fresh state and bring back headless deCONZ.

Whew.

s4y, 2017-12-02:
(I'm not closing this issue because I'm not sure what caused this to happen, so there's still potentially a bug.)

manup, 2017-12-02:
I think deCONZ, REST API and App should show a warning when a a possible invalid configuration is detected and provide guidance how to solve it.

The MAC address issue is not that common, but invalid TC link key pops up from time to time.

closed 2019-03-30 by stale[bot]