dresden-elektronik/deconz-rest-plugin #280: Hue A19 will no longer pair

s4y, 2017-11-21:
I have one Philips Hue A19 which I can no longer pair with a Raspbee + deCONZ. If I perform a Touchlink reset, the light turns off and back on (which I believe means the reset worked), but it never joins the network.

In the device scan list, it shows up on the right channel but with the network ID `0xfffe`, which isn't right (this ID is consistent). When I plugged in my old Hue v1 bridge, it immediately found and was able to control the light.

Before I moved the Raspbee to a new Pi and did some resetting (see #279), it was paired and being controlled successfully.

What can I try next?

ebaauw, 2017-11-21:
Network id 0xfffe is the default after a reset to factory new state.

You might try and power off all other lights and move the A19 in range of the RaspBee.  Select Open Network in the deCONZ web app and then power on the A19.

s4y, 2017-11-21:
Oh, that’s good to know (about the network ID).

I just tried what you suggested with no luck. The Touchlink step worked fine (the bulb acknowledged it), but it doesn’t join the network. All other bulbs were powered off.

In general, it looks like the bulb follows the bridge to the right channel but doesn’t go beyond that. And I know that it works from trying it successfully with the old bridge.

manup, 2017-11-21:
The gateway network settings might also cause this behavior, for example it is very important that the trust center link key and security settings are valid as HA coordinator.

pschlosser, 2017-11-24:
I have a similar issue with Osram Garden Spot.. It worked until I rebuild my RaspberryPi and switched to headless. I found all lights with the WebPanel but the garden spot. It reacts with touchlink (blink and reset), after opening the network it blinks again once, but it doesn't appear in the list.

Is there maybe a problem because it was allready linked to the RaspBee module before?

s4y, 2017-11-25:
@manup How can I check/troubleshoot these things?

manup, 2017-11-25:
> Is there maybe a problem because it was allready linked to the RaspBee module before?

Shouldn't be a problem, lights can be joined to other gateways after reset.

You may also try:
1. open the network in the webapp
2. execute the manual OSRAM reset sequence

[image]

ebaauw, 2017-11-25:
Note that most lights (except Philips, but I'm not sure about their latest firmware) only pair with a bridge/gateway briefly after power on.  My "best practice" is:
1. Reset the light (through its reset sequence, deCONZ touchlink, or Hue dimmer switch (press and hold On and Off for 10 seconds while holding the switch next to the light);
2. (Optional) double check the light has been reset using deCONZ touchlink (network id 0xfffe).  If needed, run your Pi from a USB powerbank to move the RaspBee close to the light (make sure to enable WiFi before disconnecting the network cable!) and power off any other lights in the vicinity;
3. Power off the light;
4. Open the network from the deCONZ web app;
5. Power on the light.

> How can I check/troubleshoot these things?

In the deCONZ GUI, press the network settings icon (next to "In Network") and check:
- Security Mode: High - No Master but TC Link Key
- TC Address: Mac address of your RaspBee/ConBee (0x00212effffxxxxxx)
- TC Link Key: 0x5a6967426565416c6c69616e63653039

pschlosser, 2017-11-25:
I tried with the light's reset sequence. The Light blinks for confirmation. Checked with deCONZ Touchlink the Network ID. It is not 0xfffe but 0xc920. When I open the network it confirms with one blink, but does still not appear in the overview. When I check again with touchlink the network ID of my RaspBee is shown.

manup, 2017-11-30:
Have you checked the network settings as suggested by ebauu too? It's very important that they are correct.

s4y, 2017-11-30:
@manup: I’m running on a headless system, is there any way to check these via the API or the web interface?

s4y, 2017-12-02:
OK, I spun up a GUI. Here's what I see. I'm happy to enter the other info, but I don't know the RaspBee's MAC address (and don't remember having to enter it before):

[image]

oleg77s, 2017-12-02:
@s4y 
As @ebaauw commented, TC Link Key: 0x5a6967426565416c6c69616e63653039, 
not a Network Key

ebaauw, 2017-12-02:
deCONZ should read the mac address of the RaspBee and show it in the GUI.  You should only need to enter it under _TC Address_.

I don’t know why it doesn’t show.  I’d try resetting the RaspBee:
- Shutdown deCONZ;
- From the command line issue: `sudo GCFFlasher_internal -r`.

Alternatively, shutdown the Raspberry and disconnect it from power.

s4y, 2017-12-02:
Unfortunately, if I enter the TC Link Key and click "Save", then click "Read", it goes back to all zeros.

Reset seems to work but doesn't make the MAC address start showing up in deCONZ (I've tried power cycling too):

```
rmmod: ERROR: Module ftdi_sio is not currently loaded
rmmod: ERROR: Module usbserial is not currently loaded
GCFFlasher V2_10 (c) dresden elektronik ingenieurtechnik gmbh 2016/09/26
no device specified use RaspBee (/dev/ttyS0)
reset target SUCCESS
```

manup, 2017-12-02:
You can also enable the `Custom Mac Address` option and set the `MAC Address` manually *and* enter the same address in `TC Address`.

Then:
* click `Save`
* close network settings
* click `Leave`
* wait 5 seconds
* click `Join`

s4y, 2017-12-02:
@manup @ebaauw Thanks both for all your help. This seems to have resolved the other issues, although I haven't been able to try pairing the missing light again because I'm not nearby (I'm doing everything else remotely and happened to have a picture of the label).

After I ran through those steps, the TC Link Key field filled itself in. I also tried *unchecking* `Custom MAC Address` and clicking `Read` (the MAC address was still there), then resetting the RaspBee and rebooting the Pi. The MAC address is still there and appears as the `bridgeid` in the REST API, which resolves #279.

It still seems like there's a bug here — any idea why this would have happened?

s4y, 2017-12-02:
(I'll be back on Sunday night and should be able to try pairing the light.)

s4y, 2017-12-04:
Yes, as soon as I opened the network and powered on the light, it joined the network! Thanks again. It looks like #279 was the cause.

closed 2017-12-04 by s4y