dresden-elektronik/deconz-rest-plugin #281: Xiaomi Aqara Motion Sensor Status

to0b, 2017-11-21:
hey,

I‘ve added a aqara motion sensor with the beta phoscon app but the sensor status is only „motion detected“. I‘ve added a rule in the group control of my connected lights and the rule works fine with no detected motion and motion. 
Also the lux value don‘t change.
Please help.

manup, 2017-11-21:
The motion detected state is cleared a while after no more motion is detected (minutes). This duration is currently not configurable via app. The light state is indeed wrong since Xiaomi uses plain lux value for the sensor, this will be fixed in the next version.

to0b, 2017-11-21:
Ok thanks! So i‘ve waited 20 minutes and the state doensn‘t changed. It‘s still detected.

manup, 2017-11-21:
Oops you're right, just testet it; somehow the `duration` attribute is not created and causes the motion state to not set back automatically.

Will be fixed in the next version

ebaauw, 2017-11-21:
I think it’s created when the sensor is restored from the database, so restarting deCONZ might help you out for now.  Be sure to set duration to something greater than 0.

Krocko, 2017-11-23:
The max lightlevel i can get is 2 lux. Also on daylight.

manup, 2017-11-23:
Do you already use latest commits from master? The fix is not in current 2.04.90 but will be available in 2.04.91 which will be online this week.

Krocko, 2017-11-23:
Yes, i use the latest commits.

to0b, 2017-11-23:
I've compiled the last commits and the lightlevel shows now 2,7 lux. The sensor is outside, so to me it seems that the value is a bit to low? My hue motion inside the house shows 7 lux.

manup, 2017-11-23:
You may try the latest version 2.04.91.
Also note that the sensor doesn't update the light sensor value periodically, it is only send when motion is detected.

pietervandersluis, 2017-11-23:
Is the other light value not more accurate?
```
{
    "config": {
        "on": true,
        "reachable": true,
        "tholddark": 12000,
        "tholdoffset": 7000
    },
    "ep": 1,
    "etag": "065391d16b8876d0d7602aeb2f5ce9d5",
    "manufacturername": "LUMI",
    "modelid": "lumi.sensor_motion.aq2",
    "name": "Hall Light Level",
    "state": {
        "dark": true,
        "daylight": false,
        "lastupdated": "2017-11-23T22:41:59",
        "lightlevel": 2, << ------------------------------------
        "lux": 1
    },
    "type": "ZHALightLevel",
    "uniqueid": "00:15:8d:00:01:65:6d:c1-01-0400"
}
```

PS: Where I live it is 23:52 so it is pretty dark now.

manup, 2017-11-23:
The value `lux` is what this sensors sends as raw unmodified value. `lightlevel` is calculated from it according to ZigBee spec.

pietervandersluis, 2017-11-23:
Also the `"modelid": "lumi.sensor_motion"` does not reset back to `False`. Not that it bothers me but just so you know :-).

And thanks again for the quick response

manup, 2017-11-23:
It's only the `presence` attribute which will turn to false when no motion is detected. Model id is static :)

to0b, 2017-11-24:
Is there a defined period of time when other sensors report their values like the xiaomi temp and hum sensor?

Krocko, 2017-11-24:
With deCONZ .91 the `luxlevel` worked.

ebaauw, 2017-11-24:
> The value `lux` is what this sensors sends as raw unmodified value. `lightlevel` is calculated from it according to ZigBee spec.

It's the other way round: the sensor reports `lightlevel` as raw value and deCONZ computes `lux` from that.  From the ZCL spec:

> 4.2.2.2.1 _MeasuredValue_ Attribute
_MeasuredValue_ represents the Illuminance in Lux (symbol lx) as follows:
_MeasuredValue_ = 10,000 x log10 Illuminance + 1
Where 1 lx <= Illuminance <=3.576 Mlx, corresponding to a _MeasuredValue_ in the range 1 to 0xfffe. The _MeasuredValue_ attribute can take the following values.
> - 0x0000 indicates a value of Illuminance that is too low to be measured.
> - MinMeasuredValue ≤ MeasuredValue ≤ MaxMeasuredValue under normal circumstanc-
es.
> - 0xffff indicates that the Illuminance measurement is invalid.

The Hue bridge only reports `lightlevel` and leaves the `lux` computation to the app.

> Is there a defined period of time when other sensors report their values like the xiaomi temp and hum sensor?

This really differs per sensor:
- Philips Hue motion sensors use ZigBee attribute reporting configuration.  Both the Hue bridge and deCONZ configure this to send the value every 5 minutes and when (significant) changes occur.  Also not that these sensors will report no motion explicitly, implementing `duration` on the sensor;
- Xiaomi sensors (at least the ones I've seen) have a hard-coded reporting configuration.  I haven't analysed them extensively, but it would seem they only send reports on (significant) changes, no regular reports.

manup, 2017-11-24:
>It's the other way round: the sensor reports lightlevel as raw value and deCONZ computes lux from that. From the ZCL spec:

Yes but in case of Xiaomi motion+light sensor it's the other way around, they are "again" doing things different here :)

ebaauw, 2017-11-24:
> Yes but in case of Xiaomi motion+light sensor it's the other way around, they are "again" doing things different here :)

Of course they are :-(

I doubt they report `lux` as raw value however.  For bright daylight, the value for`lux` would be greater than the maximum value for _Measured Value_ (0xfffe).  They probably use another formula to convert raw value to `lux`.  Does anyone have access to both deCONZ, the Xiaomi sensor, and the Xiaomi gateway and app to compare raw values (from deCONZ) against the reported values in Xiaomi app?

manup, 2017-11-24:
> I doubt they report lux as raw value however. For bright daylight, the value forlux would be greater than the maximum value for Measured Value (0xfffe). They probably use another formula to convert raw value to lux.

Yep I think that's the case, at least the values are very similar then I put the Philips hue sensor next to it. Comparison with Xiaomi gateway would be very useful indeed.

basty2, 2017-12-01:
Hi, 
i'm using deconz with xiaomi body motion sensors and it working finde. Thank you. :) 

If I pair the sensor with deconz even when I push the reset button the sensor sends a motion signal every 5 seconds if there is motion. 
After a couple of hours, the sensors sende a signal only one time a minute (maybe it changes to a battery saving mode). Can I change the duration of sending signals to the mode (5 seconds) after the reset?

manup, 2017-12-02:
Yes the 1 minute interval is on purpose to save battery life, I'm afraid that can't be changed. Every 5 seconds in a busy place would empty the battery in days :)

So the minimum duration for this specific sensor should be 60 seconds, which should be fine for most use cases I guess.

Pagani99, 2018-03-20:
Minimum duration of 60 seconds...
Could this explain the following:
I set the duration of the sensor to 15 sec. After 15 sec, the presence is false, but for some time (I guess the rest of the 60 seconds then), the sensor will not work anymore. Presence will stay false, even if you move in front of it.

On the API page, the config is showing 15 sec however:
```
duration | 15
on | true
reachable | true
```

closed 2019-03-30 by stale[bot]