dresden-elektronik/deconz-rest-plugin #282: fatal error: stdlib.h

ralphyX, 2017-11-22:
I have still a compling error using strech, the workaround [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/216#issuecomment-335021651](url) doesn't work for me

my Version:
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.1 (stretch)
Release:	9.1
Codename:	stretch

`In file included from /usr/include/c++/6/bits/stl_algo.h:59:0,
                 from /usr/include/c++/6/algorithm:62,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/qglobal.h:94,
                 from /usr/include/arm-linux-gnueabihf/qt5/QtCore/QtGlobal:1,
                 from de_web_plugin_private.h:13,
                 from atmel_wsndemo_sensor.cpp:11:
/usr/include/c++/6/cstdlib:75:25: fatal error: stdlib.h: Datei oder Verzeichnis nicht gefunden
 #include_next <stdlib.h>`

mariusmotea, 2017-11-22:
Same for me, just compile with `qmake QMAKE_CFLAGS_ISYSTEM= && make -j2`

ralphyX, 2017-11-22:
thanks, worked for me!

closed 2017-11-22 by ralphyX