dresden-elektronik/deconz-rest-plugin #283: Install Deconz to Docker?

joq3, 2017-11-22:
Hi,

I have a Debian 8 x64 server running Home Assistant via a Docker image. I am about to purchase a Conbee, and integrate that into Home Assistant. And from my understanding Deconz is required to make this work.
How do I install Deconz in a Docker container. Where do I pull the image from?
And what is required to make this work? I have to give access to the USB unit via the Docker container?

Help is appreciated.

marthoc, 2017-11-22:
@joq3 I believe we had an exchange on the Home Assistant forums earlier - I am working on a Docker image for x86_64 and armhf as we speak and have a “working” version of x86_64 that is in rough form but builds and is ready for testing.

However as I have also mentioned I am currently waiting for hardware to arrive so that I can test before releasing. I would be happy for other users interested in a Docker image to let me know and provide testing so that the image can work as well as possible. 

Dresden Elektronik does not offer an official Docker image nor do they support Docker but I expect that I will have the support of @manup to deal with any issues as they arise.

manup, 2017-11-22:
Yep I'm keen to help with deCONZ related topics, we currently don't have experience with Docker so I hope to learn a few things too :)

marthoc, 2017-11-23:
@manup - Can you explain to me how the code in this repository relates to the beta releases of deCONZ for both Ubuntu and the RPi? It appears that the deCONZ deb files already contain the REST plugin. Is it necessary to both install the deCONZ deb _and_ then compile the code in this repo, or does (for example) deconz-2.04.86-qt5.deb already contain version 2.04.86 of the REST plugin? Do you advise to keep deconz and rest plugin at the same version? :)

manup, 2017-11-23:
>Is it necessary to both install the deCONZ deb and then compile the code in this repo, or does (for example) deconz-2.04.86-qt5.deb already contain version 2.04.86 of the REST plugin? 

The plugin is always included in the deCONZ package, compilation is only needed for modification and to test bleeding edge commits from master.

See also: https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional

> Do you advise to keep deconz and rest plugin at the same version? :)

For normal users I would just recommend to use the main package of deCONZ, since it already holds close to master branch commits.

marthoc, 2017-11-23:
Thank you @manup that is very helpful.

marthoc, 2017-11-23:
Also; is there some documentation about the various command line options for deCONZ when running headless?

marthoc, 2017-11-24:
@manup I'm not sure if this is possible, but it would be extremely helpful to Docker development if, when adding Ubuntu and RPi .deb files to the website, you also added the latest version as deconz-beta-latest.deb or something similar - this way I could more easily automate the build process and easily keep the images up to date with new releases. I realize this is more work for you though and could be prone to error. 🙂

manup, 2017-11-24:
Agree, and it is quite simple too, just a symlink. I'll add that in the next days.

Currently the whole upload process is manually, the plan is to automate it completely and also provide more meta data like Debian repository, JSON file and RSS feed with version updates.

manup, 2017-11-24:
>Also; is there some documentation about the various command line options for deCONZ when running headless?

Currently these are not documented, they were requested a few times and will be added to the documentation / README.md.

Most important ones:

```
--http-port=80     http port to use
--upnp=0           disable UPnP/SSDP discovery         
--dbg-info=1       general debug informations
--dbg-info=2       to get more noise
--dbg-aps=1        ZigBee APS level logging
--dbg-zdp=1        ZigBee ZDP level logging
--dbg-zcl=1        ZigBee ZCL related logging
--dbg-otau=1       ZigBee OTA update logging
```

joq3, 2017-11-24:
Great to hear that there are plans for this! I will go ahead and purchase a Conbee.
One more thing, will it be easy to set the port for Deconz when running in a Docker container?
I already have a webserver running on port 80 so I would like to manually set the port.

marthoc, 2017-11-24:
@joq3 Yes, it will be very easy to change the port. :)

abmantis, 2017-11-25:
Not sure if you guys need it, but you can find my Dockerfile here: https://gist.github.com/abmantis/09df711ece2e9f5b5156fea193a05862

It is a bit messy ATM. I plan to improve it latter. 
For now just bump the SW_VERSION to move it forward.

marthoc, 2017-11-25:
@abmantis Thanks! I’m working on both x86 and arm Docker versions to make images that will auto deploy to Docker Hub and that will update regularly when there are new versions released. Good to see your approach.

jo-me, 2017-11-27:
Hey @abmantis , have you tried this with the ConBee stick? 

For me, the device detection did not work when Deconz is run inside a docker container. Especially, the ConBee stick is not detected. THere is a fallback to Raspbee device, but this leads to issues with firmwareupdates and empty channel ids (see other issues in this repo).

abmantis, 2017-11-27:
@jo-me Yes I tried it in the beginning for a short time. It worked, but I had to map it to AMA0. Also the firmware flashing tool didn't work.

marthoc, 2017-12-10:
@manup I have been working away on Docker since I received my testing hardware earlier this week (thank you very much again for that). I am also seeing the issue reported by other users (@jo-me and @abmantis) that the Conbee is not recognized in the container (channel and network id 0). However, interestingly, if I run the reset gateway command, the gateway appears to be initialized and receives a channel and network id. This seems to indicate to me that there is some serial port initialization performed by the reset gateway command (in addition to the resetting of the gateway, obviously). Obviously, I don’t have access to the deCONZ source to see exactly what calls are being made through QTSerialPort to make this happen, but can you share some details at a high level of what the Reset Gateway command does with the serial port? I’m thinking that all that may be required is a simple script or program that runs on Docker container startup to initialize the serial port. Thanks!

marthoc, 2017-12-20:
@manup - I know you’re busy, but wondering if you had a few moments to address my question above. Essentially, how does deconz detect which device is attached (Conbee or RaspBee) and where (which serial port for Conbee). I think this is the last major block to getting deconz working reliably in Docker, as if deconz is relying on udev then it will fail since udev does not run inside a Docker container (except with some special workarounds). Is it possible to specify the device in a command line parameter to deconz?

manup, 2017-12-25:
Hey @marthoc I'll have a look in the serial port code the next days and give you the related infos. If I remember correctly the lib is just initialized with the path to serial device. So maybe it's simplest to add a commandline parameter for that.

phithor, 2018-01-16:
Hi, have any of you managed to get a solution to this problem? I am struggling with the same

marthoc, 2018-01-16:
Hi @phithor, I have struggled with this issue for a while and I think the solution will be as @manup has suggested - adding a command line parameter to deCONZ to allow specifying the device. This is a change that needs to come from DE however as deCONZ is not open source.

nilzen, 2018-01-16:
@marthoc is the source code for your Docker image available somewhere so I can have a look?
But it sounds like, from reading this discussion, that the simplest path would be if @manup added a command line parameter for it.

A well working and maintained Docker image would be awesome for the HA community.

xdream101im, 2018-01-27:
Hi @marhoc, 
In my quest to getting deCONZ working, I came across these addon/docker built.

The built, but I can't seem to get it working with the raspbee

https://github.com/fuglphoenix/Hassio-addon-deCONZ (does not work with the latest deconz)
https://github.com/razem-io/docker-deconz

Not sure if it helps.

gjong, 2018-02-02:
@xdream101im for a hassio addon with raspbee I have created a docker image. It still needs some work, but in have got it runing and detecting zigbee devices. See https://bitbucket.org/jongsoftdev/hassio-addon/src/7e7bc50056e17fbacce26b087835e3b805392107/dresden-deconz/?at=master

doctorjames, 2018-02-02:
My [comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/236#issuecomment-361907590) on #236 might be useful to followers of this issue regarding successfully getting deConz working successfully in a container.

marthoc, 2018-03-01:
Quick update for anyone following this issue: my deCONZ Docker image now supports Conbee (and maybe RaspBee, I haven't tested yet) on amd64 and armhf: https://hub.docker.com/r/marthoc/deconz/. Feedback and any bug reports are appreciated at the image's GitHub repo: https://github.com/marthoc/docker-deconz.

kylegordon, 2018-03-20:
Looks great! Up and running in no time at all with the docker-compose.yaml file :-)

AndreasRavn, 2018-03-22:
Great work @marthoc ! ☺️

marthoc, 2018-03-28:
Another quick update - firmware updates are now possible using my image from the command line (unfortunately, not from the web UI yet, I’m working on making that more elegant): see the README at https://github.com/marthoc/docker-deconz for firmware flashing instructions. Tested and working for Conbee and RaspBee on both amd64 and armhf. Please leave any feedback or issues on my GitHub repo. Thanks!

qm3ster, 2018-11-30:
Anyone know what's actually stored to `/opt/deconz`? All the reports I read say it's empty.

closed 2019-04-06 by stale[bot]