dresden-elektronik/deconz-rest-plugin #284: Dependency with wiringpi

esseki, 2017-11-23:
Hi,

I bought a Conbee and I finally received it so I’m eager to play with it :)

I have a RPi (version 1) at home but it is already being used and I am concerned it may not be powerful enough to handle Deconz on top of its current usage. Do you know if acting as a Zigbee controller requires a lot of power or if a RPi 1 is sufficient for that ?

Anyway my plan was to plug the Conbee on a little Cubox-i Armbox that I already have. 
The problem is that when trying to install Deconz, it fails with the following missing dependency : 

 _deconz depends on wiringpi; however:
  Package wiringpi is not installed._

Of course the Cubox-i is not a Rpi and there is no GPIO on this board. So the dependency cannot be satisfied. The thing is, in my case (I bought a Conbee USB stick) this library would anyway be useless (correct me if I’m wrong).

I’m wondering if, without this dependency, your solution would not be compatible with much more hardware than just the RPi (again I’m just talking about the Conbee USB stick).

My questions : 
1) is this dependency absolutely required even in case someone wants to use a Conbee and a RPi ?
2) if the dependency is not required for Deconz to run when a Conbee is plugged to it, is there any solution you are aware of, that would allow me to install Deconz without this library ?
3) in the long term, shouldn't you publish packages for Deconz without this dependency (if it is not required) and ask your client to install the library only if they bought a Rasbee ? It would make your solution compatible with much more Arm boards (like Odroid C2 or Cubox in my case) that are all available with Debian Arm distributions.

Thank you in advance for your feedbacks.

manup, 2017-11-23:
> I have a RPi (version 1) at home but it is already being used and I am concerned it may not be powerful enough to handle Deconz on top of its current usage. Do you know if acting as a Zigbee controller requires a lot of power or if a RPi 1 is sufficient for that ?

The RPi 1 does work too, it's a bit slower but that should be ok to start with.

> deconz depends on wiringpi; however:
Package wiringpi is not installed.

deCONZ itself doesn't depend on wiringpi, only the bootloader tool GCFFlasher does, which is used for firmware updates.

So if you install deCONZ by ignoring this dependency it should work too (not testet but there should be a option/flag in apt for that purpose).

> in the long term, shouldn't you publish packages for Deconz without this dependency (if it is not required) and ask your clients to install the library only if they bought a Rasbee ? It would make your solution compatible with much more Arm boards (like Odroid C2 or Cubox in my case) that are all available with Debian Arm distributions.

The dependency was added recently to simplify installation process, so that `apt install -f` will install all needed packages. Not sure how to disable this without loosing that ability.

esseki, 2017-11-23:
Hi @manup 

Thank you for your answer.
I finally made it. The Conbee I received today is working with Deconz 2.04.91 (latest) on my cubox-i 👍 

In case it can help others, here is what I did : 

I installed all the required dependencies manually
`sudo apt install libqt5core5a libqt5network5 libqt5widgets5 libqt5gui5 libqt5serialport5 libqt5websockets5 libqt5sql5 sqlite3`

Then I installed Deconz package ignoring the wiringpi dependency
`sudo dpkg --ignore-depends wiringpi -i deconz-2.04.91-qt5.deb`

I have a last question, when you say : 
> 
deCONZ itself doesn't depend on wiringpi, only the bootloader tool GCFFlasher does, which is used for firmware updates.
> 
Does it include also firmware updates for the Conbee or just for the Raspbee ?
In case the library wiringpi is required to update the Conbee firmware too, would there be another way for me to update the Conbee firmware (using a windows or Mac OS computer) ?

Thank you for your answer

manup, 2017-11-23:
Cool glad that it works.

>Does it include also firmware updates for the Conbee or just for the Raspbee ?
In case the library wiringpi is required to update the Conbee firmware too, would there be another way for me to update the Conbee firmware (using some software from a Windows or Mac OS computer for example) ?

I hope in 2018 to create a version of GCFFlasher which won't depend on wiringpi (it is only used to toggle the reset pin anyway).

Yes GCFFlasher is also available for Windows and Ubuntu and can be used to Update the firmware of ConBee.

https://www.dresden-elektronik.de/funktechnik/service/download/driver/?L=1#c6102

esseki, 2017-11-23:
Sounds good. I will proceed with firmware updates from my Windows machine using GCFFlasher and run the Zigbee network from the Linux based Arm Cubox-i board with the Conbee attached.

Last question where can I download Conbee's firmware from ? I would like to download them to apply update from Windows manually ? And can I check from Deconz the version of the current version of the firmware installed in my Conbee ?

I would like to thank you very much for your quick answers. This is much appreciated by a new customer like me :)

manup, 2017-11-23:
Firmware files are part of the deCONZ package and on Linux they are stored in `/usr/share/deCONZ/firmware`.

You can see the current version in *deCONZ > Help > About*; and the easiest way to install new firmware is running deCONZ on Windows/Linux with attached ConBee, if a update is available the webapp will show a Update button.

esseki, 2017-11-24:
Firmware version 0x260B0500 in Deconz 

ls -la /usr/share/deCONZ/firmware
-rw-r--r-- 1 esseki esseki 129878 Aug 23 09:29 deCONZ_Rpi_0x26160500.bin.GCF
-rw-r--r-- 1 esseki esseki 130452 Sep  4 03:51 deCONZ_Rpi_0x26190500.bin.GCF

I will update tonight with GCFFlasher from a machine running Windows !

I'm good with all your answers. Closing the ticket.
Thank you !

closed 2017-11-24 by esseki

esseki, 2017-11-25:
To keep people in the same situation as me (willing to use a Conbee with another board than a RPi) updated, I could update the Conbee firmware very easily with the Windows version of GCFFlasher.

The command is well documented in Conbee PDF documentation (see chapter 5.1 Installing individual firmware with GCFFlasher in [https://www.dresden-elektronik.de/funktechnik/solutions/wireless-light-control/gateways/conbee/?eID=dam_frontend_push&docID=5042](url)).

ThePoshExpert, 2020-03-02:
I am also using an odroid on Ubuntu 18.04.2 LTS arm and got a phoscon conbee 2 running with these command nearly like User esseki did:

sudo apt install libqt5core5a libqt5network5 libqt5widgets5 libqt5gui5 libqt5serialport5 libqt5websockets5 libqt5sql5 sqlite3 **lsof**
wget https://deconz.dresden-elektronik.de/raspbian/stable/deconz-latest.deb
sudo dpkg --ignore-depends wiringpi -i deconz-latest.deb

mjtic12, 2022-11-25:
Did the same like ThePoshExpert but it does not worked properly.
I had to remove the "wiringpi" in the dependencies check in the file /usr/bin/deCONZ-autostart.sh