dresden-elektronik/deconz-rest-plugin #285: deconz keeps core dumping since .91

holli73, 2017-11-23:
```
holli@obelix:~$ /usr/bin/deCONZ
Qt: XKEYBOARD extension not present on the X server.
Segmentation fault (core dumped)
holli@obelix:~$ /usr/bin/deCONZ 
Qt: XKEYBOARD extension not present on the X server.
*** Error in `/usr/bin/deCONZ': corrupted double-linked list: 0x000000000276c230 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f27ca56e7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x82970)[0x7f27ca579970]
/lib/x86_64-linux-gnu/libc.so.6(__libc_malloc+0x54)[0x7f27ca57b184]
/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_Znwm+0x18)[0x7f27cae6de78]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QDateTimeC1Ev+0x13)[0x7f27cb46b9b3]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QDateTime19fromMSecsSinceEpochExN2Qt8TimeSpecEi+0x2f)[0x7f27cb47028f]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QDateTime15currentDateTimeEv+0x2c)[0x7f27cb4706bc]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN5QTime11currentTimeEv+0x21)[0x7f27cb470771]
/usr/share/deCONZ/plugins/libde_rest_plugin.so(_ZN12RestNodeBaseC1Ev+0x132)[0x7f27abd4b6b2]
/usr/share/deCONZ/plugins/libde_rest_plugin.so(_ZN9LightNodeC1Ev+0x2e)[0x7f27abc8d00e]
/usr/share/deCONZ/plugins/libde_rest_plugin.so(_ZN19DeRestPluginPrivate12addLightNodeEPKN6deCONZ4NodeE+0xc8)[0x7f27abc5e6d8]
/usr/share/deCONZ/plugins/libde_rest_plugin.so(_ZN19DeRestPluginPrivate9nodeEventERKN6deCONZ9NodeEventE+0x3b3)[0x7f27abc5fef3]
/usr/share/deCONZ/plugins/libde_rest_plugin.so(+0x1cc0bc)[0x7f27abda40bc]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN11QMetaObject8activateEP7QObjectiiPPv+0x7ea)[0x7f27cb633d2a]
/usr/lib/libdeCONZ.so.1(_ZN6deCONZ13ApsController9nodeEventERKNS_9NodeEventE+0x35)[0x7f27cc466e95]
/usr/bin/deCONZ[0x460b66]
/usr/bin/deCONZ[0x463170]
/usr/bin/deCONZ[0x4647cd]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN7QObject5eventEP6QEvent+0xa3)[0x7f27cb634bb3]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN19QApplicationPrivate13notify_helperEP7QObjectP6QEvent+0x8c)[0x7f27cbef805c]
/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5(_ZN12QApplication6notifyEP7QObjectP6QEvent+0x256)[0x7f27cbefd516]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication14notifyInternalEP7QObjectP6QEvent+0xdb)[0x7f27cb60538b]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN14QTimerInfoList14activateTimersEv+0x52d)[0x7f27cb65a5ed]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2dbaf1)[0x7f27cb65aaf1]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_dispatch+0x2a7)[0x7f27c9fdd197]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(+0x4a3f0)[0x7f27c9fdd3f0]
/lib/x86_64-linux-gnu/libglib-2.0.so.0(g_main_context_iteration+0x2c)[0x7f27c9fdd49c]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN20QEventDispatcherGlib13processEventsE6QFlagsIN10QEventLoop17ProcessEventsFlagEE+0x5f)[0x7f27cb65b7cf]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN10QEventLoop4execE6QFlagsINS_17ProcessEventsFlagEE+0x10a)[0x7f27cb602b4a]
/usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN16QCoreApplication4execEv+0x9c)[0x7f27cb60abec]
/usr/bin/deCONZ[0x430e86]
/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf0)[0x7f27ca517830]
/usr/bin/deCONZ[0x431379]
======= Memory map: ========
00400000-0051c000 r-xp 00000000 fc:00 25430814                           /usr/bin/deCONZ
0071b000-00721000 r--p 0011b000 fc:00 25430814                           /usr/bin/deCONZ
00721000-00724000 rw-p 00121000 fc:00 25430814                           /usr/bin/deCONZ
00724000-00725000 rw-p 00000000 00:00 0 
01e66000-02b4d000 rw-p 00000000 00:00 0                                  [heap]
7f2790000000-7f2790021000 rw-p 00000000 00:00 0 
7f2790021000-7f2794000000 ---p 00000000 00:00 0 
7f2794000000-7f2794021000 rw-p 00000000 00:00 0 
7f2794021000-7f2798000000 ---p 00000000 00:00 0 
7f279b5f8000-7f279b5f9000 ---p 00000000 00:00 0 
7f279b5f9000-7f279bdf9000 rw-p 00000000 00:00 0 
7f279bdf9000-7f279bdfe000 r-xp 00000000 fc:00 5902370                    /lib/x86_64-linux-gnu/libnss_dns-2.23.so
7f279bdfe000-7f279bffe000 ---p 00005000 fc:00 5902370                    /lib/x86_64-linux-gnu/libnss_dns-2.23.so
7f279bffe000-7f279bfff000 r--p 00005000 fc:00 5902370                    /lib/x86_64-linux-gnu/libnss_dns-2.23.so
7f279bfff000-7f279c000000 rw-p 00006000 fc:00 5902370                    /lib/x86_64-linux-gnu/libnss_dns-2.23.so
7f279c000000-7f279c2ac000 rw-p 00000000 00:00 0 
7f279c2ac000-7f27a0000000 ---p 00000000 00:00 0 
7f27a0000000-7f27a0021000 rw-p 00000000 00:00 0 
7f27a0021000-7f27a4000000 ---p 00000000 00:00 0 
7f27a4000000-7f27a4021000 rw-p 00000000 00:00 0 
7f27a4021000-7f27a8000000 ---p 00000000 00:00 0 
7f27a8086000-7f27a8088000 r-xp 00000000 fc:00 5898284                    /lib/x86_64-linux-gnu/libnss_mdns4_minimal.so.2
7f27a8088000-7f27a8287000 ---p 00002000 fc:00 5898284                    /lib/x86_64-linux-gnu/libnss_mdns4_minimal.so.2
7f27a8287000-7f27a8288000 r--p 00001000 fc:00 5898284                    /lib/x86_64-linux-gnu/libnss_mdns4_minimal.so.2
7f27a8288000-7f27a8289000 rw-p 00002000 fc:00 5898284                    /lib/x86_64-linux-gnu/libnss_mdns4_minimal.so.2
7f27a8289000-7f27a82a0000 r-xp 00000000 fc:00 5902515                    /lib/x86_64-linux-gnu/libresolv-2.23.so
7f27a82a0000-7f27a84a0000 ---p 00017000 fc:00 5902515                    /lib/x86_64-linux-gnu/libresolv-2.23.so
7f27a84a0000-7f27a84a1000 r--p 00017000 fc:00 5902515                    /lib/x86_64-linux-gnu/libresolv-2.23.so
[... 150 lines cut ...]
7f27bcefb000-7f27bcf09000 r-xp 00000000 fc:00 25435813                   /usr/lib/x86_64-linux-gnu/libsensors.so.4.4.0
7f27bcf09000-7f27bd108000 ---p 0000e000 fc:00 25435813                   /usr/lib/x86_64-linux-gnu/libsensors.so.4.4.0
7f27bd108000-7f27bd109000 r--p 0000d000 fc:00 25435813                   /usr/lib/x86_64-linux-gnu/libsensors.so.4.4.0
7f27bd109000-7f27bd10a000 rw-p 0000e000 fc:00 25435813                   /usr/lib/x86_64-linux-gnu/libsensors.so.4.4.0
7f27bd10a000-7f27bda97000 r-xp 00000000 fc:00 25428983                   /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
7f27bda97000-7f27bdc97000 ---p 0098d000 fc:00 25428983                   /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
7f27bdc97000-7f27bdcd2000 r--p 0098d000 fc:00 25428983                   /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
7f27bdcd2000-7f27bdcde000 rw-p 009c8000 fc:00 25428983                   /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so
7f27bdcde000-7f27bded5000 rw-p 00000000 00:00 0 
7f27bded5000-7f27bdefc000 r-xp 00000000 fc:00 25428085                   /usr/lib/x86_64-linux-gnu/libkj-0.5.3.so
7f27bdefc000-7f27be0fc000 ---p 00027000 fc:00 25428085                   /usr/lib/x86_64-linux-gnu/libkj-0.5.3.so
7f27be0fc000-7f27be0fd000 r--p 00027000 fc:00 25428085                   /usr/lib/x86_64-linux-gnu/libkj-0.5.3.so
7f27be0fd000-7f27be0fe000 rw-p 00028000 fc:00 25428085                   /usr/lib/x86_64-linux-gnu/libkj-0.5.3.so
7f27be0fe000-7f27be115000 r-xp 00000000 fc:00 25428282                   /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7f27be115000-7f27be314000 ---p 00017000 fc:00 25428282                   /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7f27be314000-7f27be315000 r--p 00016000 fc:00 25428282                   /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7f27be315000-7f27be316000 rw-p 00017000 fc:00 25428282                   /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0
7f27be316000-7f27be346000 r-xp 00000000 fc:00 25433413                   /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1
7f27be346000-7f27be545000 ---p 00030000 fc:00 25433413                   /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1
7f27be545000-7f27be546000 r--p 0002f000 fc:00 25433413                   /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1
7f27be546000-7f27be547000 rw-p 00030000 fc:00 25433413                   /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1Aborted (core dumped)
holli@obelix:~$ ^C
h

```

manup, 2017-11-23:
Is this vanilla deCONZ or self compiled plugin? My Ubuntu box runs fine (vanilla).

Can you provide the coredump file?

holli73, 2017-11-23:
hello,

i did compile on my own - but was missing .91 dev package - so i could not get my self compiled running - luckely i saw new .deb - but the installed .91.dev.deb + .91.deb deconz still failing with coredump.

the compressed coredump is still 75mb and can not be uploaded here (limit 10mb).

i only could find only one core dump - so i'm not sure for what binary.

further i do run /monitor the deconz manualy so what is triggering '/usr/bin/deCONZ-autostart.sh'

maybe the dumps are related because my monitoring saw the process is down and started it and autostart stared it again - or the other way around.

because deconz was listening on port 80 instead of 8080 but my monitoring would use following cmd:

```
...
echo "&PROCTOMONITOR# is OFFLINE - restarting"
    export DISPLAY=:1.0
    deCONZ --auto-connect=1 --dbg-error=1 --http-port=8080 --pid-file=/run/user/1000/deconz/deconz.pid &
...
```

i could provide the file if still interesed through my cloud or http download - just let me know and put a link here.

thanks
holli

manup, 2017-11-23:
From the trace it looks like the fault is happening somewhere in the LightNode constructor by settings up the ResouceItems. Currently I don't know where problem is exactly. So as next step can you please try to rebuild with doing a `make clean` before, or try the unmodified plugin.

holli73, 2017-11-23:
hello,

this is done already:

```
holli@obelix:~/BUILD/deconz$ cat make-deconz.sh
cd deconz-rest-plugin
git pull
make clean
make -j9
echo "*** sudo cp libde_rest_plugin.so /usr/share/deCONZ/plugins/libde_rest_plugin.so ***"
```

i just did a rebuild and it is failing every few minutes/seconds...

core dump available here because it is failing now every seconds and therefore the file is not too big anymore...

[deconz.91.core.dump.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/1500285/deconz.91.core.dump.zip)

manup, 2017-11-23:
Thanks I'll have a look now little late night debugging for the fun of it :)

holli73, 2017-11-23:
just to check i did install .86 again (was running before)  and it is not core dumping.

PS: **gg ** luckily my wife and kids are sleeping - so no complaints during this outages...

manup, 2017-11-23:
The coredump didn't show new insights yet.

Hmm maybe my Ubuntu box (VM, 16.04, Qt 5.5.1, G++ 5.4.0) is *a bit* out of date had only 25 MB left on disk. I resized  the partition now and will build a new package with updated system and Qt5 libs during the day.

snozzlebert, 2017-11-24:
After upgrading to .91 I had some crashes first (I don't have logs/dumps). After restarting a few times deCONZ didn't crash anymore, however almost all lights (except my Philips extended color lights) reported ```"type": "Unknown"``` in the rest api.
Reverted back to .90 and types are reported again (e.g. ```"type": "Dimmable light"```)

manup, 2017-11-24:
There seems to be some serious memory issues currently in at least the Ubuntu version. Debugging all day, fixed some via valgrind bit still deCONZ keeps crashing after short time.

@snozzlebert do you refer to Ubuntu or Raspbian version?

snozzlebert, 2017-11-24:
Raspbian, sorry to use this issue. But maybe it is related.

manup, 2017-11-24:
Seems to be related, never the less I think the problem is found. Classic C++ memory corruption :) these never get boring ... :)

manup, 2017-11-24:
2.04.92 is up; should fix the issue.

holli73, 2017-11-24:
hello,

house is on .92 - no abort so far.

thanks
holli

closed 2017-11-24 by holli73