dresden-elektronik/deconz-rest-plugin #286: Connecting centralite sensors

steveStavros, 2017-11-24:
Hi everyone!!

Has anyone tried to connect to deCONZ centralite sensors??
I was able to connect to deconz the "3-Series Micro Motion Sensor" of centralite and use it thought the rest-api by adding { VENDOR_CENTRALITE, "3328-G", emberMacPrefix } in the ‘de_web_plugin.cpp’ file.

The problem is that I tried to also pair the "3-Series Temp & Humidity Sensor" of centralite but I cannot even added to the the zigbee network (It is not displayed as a node when I follow the  join network steps).

Has anyone been successful used this sensor

Sorry for the issue. Not sure if this is the right place to post this

manup, 2017-12-03:
I've running a motion sensor from OSRAM which shows as CentraLite `Motion Sensor-A`. It wasn't easy to add it to the network. Though the setup button worked well at the first time, in further tests to reset and join the device to the network it was very unpredictable.

oleg77s, 2017-12-03:
Does it not help?
How to reset lightly (CentraLite) motion sensor:
1. Eject the battery tray and remove the battery
2. Press down the device reset button
3. While holding down the reset button, reinsert battery in the tray to factory reset the device

soinipe, 2018-01-17:
Regarding Centralite 3328-G (3328-Geu) Micro-Motion Sensor:

HW: RaspPi Jessie 4.9.35-v7+ 
SW: deCONZ 2.04.99 + [the change](https://github.com/soinipe/deconz-rest-plugin/commit/c060c64f91a4a2dfa03ad316f6eb3a2f5acebbae)

-I'm able to pair the device to network
-After pairing the device can be seen with GET /sensors, for example:
``` 
    "6": {
        "config": {
            "duration": 30,
            "on": true,
            "reachable": true
        },
        "ep": 2,
        "etag": "08530dfdf8194061775d2577492ae5d8",
        "manufacturername": "CentraLite",
        "modelid": "3328-G",
        "name": "Presence 6",
        "state": {},
        "type": "ZHAPresence",
        "uniqueid": "00:0d:6f:00:0b:bc:25:1e-02-0406"
    }
```
-However, the state does not get updated
-Update is happening only after pressing "Read" in "Cluster Info" / "Occupancy Sensing" in deCONZ UI's, example result:
```
    "state": {
        "lastupdated": "2018-01-17T20:47:11",
        "presence": false
    },
```

I'm not able to spot from traces (--dbg-info) anything suspicious. Looks like device events are not simply created - and received in DeRestPluginPrivate::handleSensorEvent(). Any idea where to debug this? For reference there is Philips Motion Sensor SML001, works ok.

soinipe, 2018-01-22:
I'm thinking found the reason - perhaps it's obvious to those who have been working with the deCONZ for a longer time. I was missing the manual "binding" explained in [issue 340](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/340) by @ebaauw. 

For automatic binding there should be at least whitelisting of "3328-G" in DeRestPluginPrivate::checkSensorBindingsForAttributeReporting(Sensor *sensor), adding that, seems working without manual binding now. 

I'm still having some issues after cold start: there is no usual "green line" between deCONZ gateway and Centralite node in the deCONZ UI - and sometimes there is red communication blink in "Presence 6", most of the blinking is blue, though. I'm not sure if still missing something in code or something weird when starting over in my environment: I tried to always clean "sensors" table but not sure if there is something else which needs initialization when re-pairing the device.

manup, 2018-02-11:
Cross reference: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/404

closed 2019-03-30 by stale[bot]