dresden-elektronik/deconz-rest-plugin #288: Type of lights are Unknown

snozzlebert, 2017-11-24:
Crashes are fixed in .92, however all lights are reported as ```"type": "Unknown"``` in the rest api.
Reverted back to .90 and types are reported again (e.g. ```"type": "Dimmable light"```).

ebaauw, 2017-11-24:
Indeed, with 2.04.92, type is Unknown for all my lights (and plugs) as well.

manup, 2017-11-24:
Hmm strange, will be fixed during the day; `type` is now a ResourceItem `RAttrType`, seems that I missed some spots during the transition.

ebaauw, 2017-11-25:
Not a spot, a level of nesting, see PR #289.

manup, 2017-11-25:
Nice, thanks for the fix :)

snozzlebert, 2017-11-26:
Fixed in 2.04.93, thanks.

closed 2017-11-26 by snozzlebert

wvuyk, 2017-11-28:
Can 2.04.93 also be compiled for the ConBee Windows environment?

manup, 2017-11-28:
Yes, sorry for the delay it's uploaded now.

https://www.dresden-elektronik.de/deconz/win/deCONZ_Setup_Win32_V2_04_93.exe