dresden-elektronik/deconz-rest-plugin #290: POST /api/username/config/restartapp no longer works

ebaauw, 2017-11-25:
In .92, doing a POST to `/api/<usernaem>/config/restartapp` still returns `[{"success":{"/config/restartapp":true}}]`, but deCONZ is not longer restarted.

manup, 2017-11-25:
Is deCONZ running headless as systemd service? 

The gui version currently doesn't restart, will be fixed in the next versions.

ebaauw, 2017-11-25:
No, I'm still on the GUI version.  It used to work some versions ago.  I only use restartapp in my scripts to re-create the gateway rules (so that the rules are re-created from `/rules/1`).  I don't do this regularly, so I don't know which version or commit broke it.

manup, 2017-11-25:
Must have happened recently while we worked on the systemd deconz-update.service script.

ebaauw, 2017-12-15:
Seems to be working in v2.04.99.

closed 2017-12-15 by ebaauw