dresden-elektronik/deconz-rest-plugin #291: Unexpected duration on CLIPPresence

snozzlebert, 2017-11-25:
I did't expect a change to CLIPPresence. Is it intentional that there is a ```duration``` attribute on the configuration of the CLIPPresence sensor? I manage the state of my CLIPPresence sensor externally using the rest api (setting the ```presence``` to ```true``` and ```false```). In older versions this worked fine. In the new version, deCONZ changes the state to ```false``` after 60 seconds. I don't want/expect that, because I think that CLIP sensors should only be changed by rules, schedules or rest api access.
If it is intentional I could of course change the sensor to a CLIPGenericFlag to solve this.

```
{
        "config": {
            "duration": 60,
            "on": true,
            "reachable": true
        },
        "etag": "1a1b3c1f19be5e44286df6cf6e446cbe",
        "manufacturername": "X",
        "modelid": "CLIPPresence",
        "name": "Home",
        "state": {
            "lastupdated": "2017-11-25T13:25:13",
            "presence": false
        },
        "swversion": "1.0.0",
        "type": "CLIPPresence",
        "uniqueid": "home"
    }
```

ebaauw, 2017-11-25:
It's intentional.  If you set `config.duration` to `0`, deCONZ shouldn't reset `state.presence` to `false`.

snozzlebert, 2017-11-25:
Thanks, that did the trick. I think 0 should be the default when migrating an old CLIPPresence sensors to one with a duration config.

ebaauw, 2017-11-28:
This change was introduced in https://github.com/dresden-elektronik/deconz-rest-plugin/commit/e00b544d90bea8d361ab16597db4500f6f38217b#diff-cfbf88092dc1a3e8571b1101c1a4b01a, I think because of #281.

The value for RConfigDuration is a UInt16 and 0 is actually a valid value, so that whole change in database.cpp doesn't make sense to me.

Adding it to de_web_plugin.cpp does, but I think maybe it should be initialised to 0 instead of 60.

manup, 2017-11-28:
The change to database.cpp was made since some real sensors didn't had a config/duration attribute, which should be added with a default value here.

I'll change that so that CLIPPresence won't have config/duration added on database loading with a value other than 0.

ebaauw, 2017-11-28:
The problem was that `duration` wasn’t added when the sensor was created.  The change to de_web_plugin.cpp solved that.   It was already added when the sensor was restored from the database: https://github.com/dresden-elektronik/deconz-rest-plugin/blob/7fdcfdea65b90d048018adff6929b3a890c73808/database.cpp#L1832

closed 2017-12-06 by snozzlebert