dresden-elektronik/deconz-rest-plugin #292: Xiaomi Temperature/Pressure/Humidity misbehaves at negative temperatures

markuspf, 2017-11-25:
I have one of the Xiaomi sensors outside (whether this is advisable or not, I do not care about very much), last night the temperatures dipped below freezing (it was about -2) which resulted in a reading of 625 degrees reported by the beta app.

I am currently not recording readings, so I'd have to report later if readings from the sensor are sought when its cold.

ebaauw, 2017-11-25:
> last night the temperatures dipped below freezing

The sensor uses a signed 16-bit integer for the temperature (in centi-degrees Celsius), so it should support sub-zero temperatures.

> I have one of the Xiaomi sensors outside (whether this is advisable or not, I do not care about very much)

I have a Xiaomi Aqara weather sensor on my balcony, but winter has yet to come here.  Lowest temperature so far is 5.3 degrees.

> I am currently not recording readings

I run [`dc_eventlog`](https://github.com/ebaauw/dc_eventlog) in the background to log any web socket notifications.
```bash
nohup dc_eventlog >dc_event.log 2>&1 &
```

Krocko, 2017-11-29:
Today i had the same problem. 
[image]

ebaauw, 2017-11-29:
Could you check the value reported in the deCONZ GUI?  And in the web socket or REST API?  And in the homebridge log?

I suspect somewhere down the line, the value is interpretated as an unsinged int16 (0xffff = 65536 = 655.36 degrees) instead of a signed int16 (0xffff = -1 = -0.01 degrees).

ebaauw, 2017-11-29:
Just tested with a CLIPTemperature sensor: setting `state.temperature` to -200 works as intended: Home shows -2°C.

Krocko, 2017-11-29:
I can look into it this evening, or tomorrow Morning. Now i am on +2*C. 
Homebridge log shows
`Nov 29 07:37:23 raspberrypi homebridge[25305]: [2017-11-29 07:37:23] [Philips Hue] Temperatursensor außen: set homekit temperature from 655.3°C to 0.5°C`

Krocko, 2017-11-29:
deCONZ GUI:

[image]

REST:

[image]

ebaauw, 2017-11-29:
> I have testet in the freezer

Brilliant!

> I suspect somewhere down the line, the value is interpretated as an unsinged int16 (0xffff = 65536 = 655.36 degrees) instead of a signed int16 (0xffff = -1 = -0.01 degrees).

Almost.  The value is interpreted as a singed int32, but same net effect.  After rectifying that, my freezer cooled down from 636°C to -19°C.  PR #302.

closed 2018-11-23 by manup