dresden-elektronik/deconz-rest-plugin #294: Osram Smart+ Mini Switch and deCONZ

oleg77s, 2017-11-26:
Just added a new Osram Smart+ Mini Switch to the deCONZ network.
[image]
And it's working out of the box: switch controls all the lights and plug in the network and it's impossible  manually configure each light, as in the case with DE switches, all switch settings are factory default
[image]
Exactly the same switch behaviour is in the Philips Hue bridge network and besides, switch is invisible in hue bridge network, as DE ones.
Is it possible to add support for this switch to deCONZ (and homebridge-hue) 
to be able to configure it ?
Regards

P.S. Switch has a touchlink function:
[image]
[image]

manup, 2017-11-26:
Cool, part of it can be implemented similar to IKEA remote control where the incoming group-/broadcasts are mapped to REST API button events. Question is if it is possible to to configure the switch to not reach all lights but only a group.

oleg77s, 2017-11-26:
It's possible in native Osram app and Osram gateway network, but how to do this in deCONZ network?

manup, 2017-11-26:
Guess we need to sniff the traffic to know what commands the OSRAM gateway sends, but good that it is possible at least.

alvar0rodrig0, 2018-11-28:
Can I add something else to this issue to request the support of this switch? I'm also looking forward to this switch to be added to deconz.
The cluster info is already published and it is the same that I have, except for the Power Source that I have it as unknown and the SW Build ID is empty in my case.

cristimi, 2019-02-12:
I have two of this switches as well and I am interested of using them with deconz.

Philje123, 2019-03-23:
I've just acquired one of these cheaply.

Just updating the software on it now and gonna give it a go.

redlegoman, 2019-07-07:
> I've just acquired one of these cheaply.
> 
> Just updating the software on it now and gonna give it a go.

How did it go?

Philje123, 2019-07-07:
No change. It still just switches all lights on and off.

olicooper, 2019-08-09:
Is there anything currently in development for this?

jamestutton, 2019-08-21:
If it helps someone fix it in deconz it has been sorted in zigbee2mqtt 

https://github.com/Koenkk/zigbee-shepherd-converters/commit/ad2d5aab2c541948da965874a219922433766f00

weltmeyer, 2019-10-03:
I would also like to see this switch working with deconz...

olicooper, 2020-01-31:
Not stale..

manueligno78, 2020-02-14:
Please support this device

olicooper, 2020-05-28:
Me and @Smanar are working on adding support for the device in the REST API (we can't add support in Phoscon as it is not open source). So far I have got the buttons to output API updates but it is not finished yet. I can't guarantee support as that is up to the repo maintainers if they merge our pull requests and also if there are no hidden issues we haven't discovered yet.
Fingers crossed we can add some form of support! 🤞

closed 2020-06-27 by stale[bot]

Fabian7727, 2020-07-06:
How do you even add it to the network. It’s simply not working at all in my system.

olicooper, 2020-07-06:
Until this PR is closed https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2917 you can't use it unless you build your own version from the PR code

Smanar, 2020-07-06:
Or if all is fine, you will it have in future deconz version in 10 days.