dresden-elektronik/deconz-rest-plugin #295: How to specify command line parameters to deCONZ v2.04.93?

ebaauw, 2017-11-26:
I used to start deCONZ by setting `DECONZ_OPTS` and then running `deCONZ-autostart.sh` in background, redirecting stdout and stderr.  As of .93, `deCONZ-autostart.sh`starts the `deconz-gui` system service and exits instead of starting the `deCONZ` executable.

It seems the output of the service is send to syslog alright:
```
Nov 26 18:35:29 pi systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Nov 26 18:35:29 pi systemd[1]: Started deCONZ: ZigBee gateway -- Initialisation.
Nov 26 18:35:29 pi deCONZ[31623]: libEGL warning: DRI2: failed to authenticate
Nov 26 18:35:29 pi deCONZ[31623]: libpng warning: iCCP: known incorrect sRGB profile
```
However the logging options in `DECONZ_OPTS` seem to be ignored.

ebaauw, 2017-11-26:
Probably the best way to do this is by providing a `/etc/default/deconz` file which sets
```
DECONZ_OPTS="--http-port=80"
```
Then change the `ExecStart` line in `/etc/systemd/system/deconz.service` to
```
EnvironmentFile=/etc/default/deconz
ExecStart=/usr/bin/deCONZ -platform minimal ${DECONZ_OPTS}
```
and in `deconz-gui.service` to
```
EnvironmentFile=/etc/default/deconz
ExecStart=/usr/bin/deCONZ ${DECONZ_OPTS}
```

We could even document all the available command line options in comments in `/etc/default/deconz`.

I'd create a PR if these files were included in the REST API repository...

manup, 2017-11-27:
Or alternatively I would recommend to use a custom systemd .service file based on the deconz.service or deconz-gui.service, with adjusted parameters. If I recall correctly there is also an systemd overlay mechanism for that purpose to extend existing service files with extra stuff.

> I'd create a PR if these files were included in the REST API repository...

Yes currently these are only part of the deCONZ package. We can transfer and check-in them to the rest plugin repository for better maintenance and PR support.

snozzlebert, 2017-11-28:
> If I recall correctly there is also an systemd overlay mechanism for that purpose to extend existing service files with extra stuff

Yes, you can create a directory ```/etc/systemd/system/deconz.service.d/``` or for deconz.gui
 ```/etc/systemd/system/deconz-gui.service.d/```
with a conf file (e.g. ```/etc/systemd/system/deconz.service.d/override.conf```)
In override.conf you can make changes:
```INI
[Service]
ExecStart=
ExecStart=/usr/bin/deCONZ --http-port=80 --ws-port=8080
```
 (```ExecStart=``` is required to clear the original value first.)

manup, 2017-11-28:
Cool, I think this is a good point to put in the documentation.

ebaauw, 2017-12-15:
I cannot seem to catch the log output when running deCONZ as a service - it's not written to the journal.  I noticed when starting deCONZ manually, the logs (for e.g. `--dbg-info=2`) are written to stdout, whereas errors (like _libEGL warning: DRI2: failed to authenticate_) are written to stderr.  I find that stdout of other services is written to the journal as expected.  Is deCONZ suppressing log messages when running as a service?

manup, 2017-12-15:
When the `--dbg-*` parameters are given in the ExecStart= field the output should be visible in systemd journal.

I'm often using `sudo journal -xb -f` to observe debug output.

ebaauw, 2017-12-16:
Odd, it works when specifying:
```
ExecStart=/usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-aps=1
```
but it doesn't work when specifying:
```
Environment="DECONZ_OPTS=--http-port=80 --dbg-info=2 --dbg-aps=1"
ExecStart=/usr/bin/deCONZ ${DECONZ_OPTS}
```
Apparently the whole string `--http-port=80 --dbg-info=2 --dbg-aps=1` is treated as a single argument, so deCONZ doesn't recognise this.  Luckily, when specifying:
```
Environment="DECONZ_OPTS=--http-port=80 --dbg-info=2 --dbg-aps=1"
ExecStart=/usr/bin/deCONZ $DECONZ_OPTS
```
(without the curly braces) it works.

manup, 2017-12-16:
The DECONZ_OPTS was only used in former deCONZ-autostart.sh script. The systemd service directly executes the /usr/bin/deCONZ binary (which was formerly done by the autostart script).

The deCONZ binary only expects the parameters as normal command line arguments.

Hence `ExecStart=/usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-aps=1` is the correct way to specify the parameters.

Early 2018 will bring a comprehensive documentation update of the REST API where these details will also be described. Kind of sad that currently documentation lacks behind all the nice stuff.

ebaauw, 2017-12-16:
> Hence ExecStart=/usr/bin/deCONZ --http-port=80 --dbg-info=2 --dbg-aps=1 is the correct way to specify the parameters.

But that will be overwritten when installing the next version of deCONZ.  Also, the system-specific options should be specified in a configuration file in `/etc/default`, not in the script in `/etc/systemd/system`.

I think I figured it out, with @snozzlebert's suggestion:

Create `/etc/default/deconz` to hold the options:
```
DECONZ_OPTS="--http-port=80 --dbg-info=2 --dbg-aps=1"
```

Create `/etc/systemd/system/deconz-gui.service.d/override.conf`:
```
[Service]
EnvironmentFile=/etc/default/deconz
ExecStart=
ExecStart=/usr/bin/deCONZ $DECONZ_OPTS
```
and `/etc/systemd/system/deconz.service.d/override.conf`:
```
[Service]
EnvironmentFile=/etc/default/deconz
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal $DECONZ_OPTS
```

manup, 2017-12-16:
Cool the EnvironmentFile is really useful. The EnvironmentFile could also be integrated into the /etc/systemd/system/deconz*.service directly so that a override service isn't needed.

Not sure if it is the proper way but DECONZ_OPTS could be splittet up into separate environment variables too, so that the config file has a more readable style, where one line refers to one variable.

**deconz-gui.service**
```
[Unit]
Description=deCONZ: ZigBee gateway -- GUI/REST API
Wants=deconz-init.service deconz-update.service
After=lightdm.service vncserver-x11-serviced.service

[Service]
User=pi
Environment="DISPLAY=:0" 
Environment="DECONZ_HTTP_PORT=80"
Environment="DECONZ_DBG_INFO=0"
Environment="DECONZ_DBG_APS=0"
Environment="DECONZ_DBG_ZCL=0"
Environment="DECONZ_DBG_ZDP=0"

# environment file, if exist, can overwrite specific variables
EnvironmentFile=-/etc/default/deconz

ExecStart=/usr/bin/deCONZ \
               --http-port=$DECONZ_HTTP_PORT \
               --dbg-info=$DECONZ_DBG_INFO \
               --dbg-aps=$DECONZ_DBG_APS \
               --dbg-zcl=$DECONZ_DBG_ZCL \
               --dbg-zdp=$DECONZ_DBG_ZDP
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target

```

Later on reading the DECONZ_* environment variables could also be integrated into deCONZ directly.

ebaauw, 2017-12-16:
> The EnvironmentFile could also be integrated into the /etc/systemd/system/deconz*.service directly so that a override service isn't needed.

Yes, please!

> Not sure if it is the proper way but DECONZ_OPTS could be splittet up into separate environment variables too, so that the config file has a more readable style, where one line refers to one variable.

Brilliant, less error prone.  We could provide an example `/etc/default/deconz` with comments per variable as documentation.  Definitely add `DECONZ_WS_PORT`.  Would we need `DECONZ_AUTO_CONNECT` as well?

kernehed, 2017-12-27:
Hi,

Is it possible to change port on deconz?

$ sudo netstat -pant|egrep "\.0:80\s|\.0:443\s"
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1029/deCONZ
tcp        0      0 0.0.0.0:443             0.0.0.0:*               LISTEN      1029/deCONZ

I would like to have some of these ports available for my Home Assistant's SSL cert.

closed 2018-06-12 by ebaauw

mansdahlstrom1, 2018-08-06:
Hi, posted in https://github.com/dresden-elektronik/phoscon-app-beta/issues/26 but saw in the readme of https://github.com/dresden-elektronik/phoscon-app-beta/ that discussion should be in this repo instead! Also since i used this Issue as a reference for my setup it seemed like a good idea to try to comment here. 

POST:  
Just tried to switch the service port using sudo systemctl edit deconz. After a bit of trouble my service seems to be running fine again. According to sudo service deconz status... and `journalctl -u deconz` logs that my connected light are being polled.

Output from sudo systemctl cat deconz:

# /etc/systemd/system/deconz.service.d/override.conf
[Service]
ExecStart=
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8080 --ws-port=8081 --auto-connect=1 --dbg-error=1
Now i can't seem to access the Phoscon app at port 8080. accessing via http://<ip-address>:8080. Can someone maybe point me in the right direction?

http://www.dresden-elektronik.de/discover/ Finds my gateway on the correct port (8080). Am I missing something here? Is the Phoscon app not part of the deconz service? Is the platform minimal disabling phoscon app.

Im running deconz-2.05.31-qt5 on a headless ubuntu machine

Thanks!

Spanishu, 2020-06-23:
I have installed my deconz follwoing the standard instructions in march 2020. 
I do not have any of those files in /etc/systemd/system
but all in /lib/systemd/system

I have tried to start the deconz app on raspbian this is the message I get back 
` libEGL warning: DRI2: failed to authenticate
Jun 23 09:45:00 homebridge deCONZ[3631]: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-pi'
Jun 23 09:45:00 homebridge deCONZ[3631]: libpng warning: iCCP: known incorrect sRGB profile`

Any suggestion on how to apply the above suggested with the different directory structure

SwoopX, 2020-06-23:
Could you please file a seperate issue? It will help keeping things clear as you seem to have deviating prerequisites. Thanks!

StephanStS, 2024-01-08:
Another option would be to use deCONZ command line parameters described there: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-command-line-parameters