dresden-elektronik/deconz-rest-plugin #298: Philips color lights have a type "unknown" in REST Api

wvuyk, 2017-11-28:
I was testing tonight with a Philips Hue Color light as I got reports my plugin was no longer creatig the color possiblilities. If I add a new Philips Hue Color light to deCONZ, the REST API reports the type as "unknown"! Is this a new bug sneaking in, or is this a consequence of the updates Philips is pushing to the lights?

Here is the REST API response:

  "6": {
      "ctmax": 500,
      "ctmin": 153,
      "etag": "55267060648e5f8b523bab0ef3ee461b",
      "hascolor": true,
      "manufacturername": "Philips",
      "modelid": "LCT010",
      "name": "Light 6",
      "state": {
        "alert": "none",
        "bri": 66,
        "colormode": "xy",
        "ct": 281,
        "effect": "none",
        "hue": 55960,
        "on": true,
        "reachable": true,
        "sat": 184,
        "xy": [
          0.40474,
          0.229936
        ]
      },
      "swversion": "1.29.0_r21169",
      "type": "Unknown",
      "uniqueid": "00:17:88:01:02:75:c3:fd-0b"
    }

snozzlebert, 2017-11-28:
Are you on 2.04.93? I had the same issue on .92 see #288

wvuyk, 2017-11-28:
No I am on ConBee 2.04.92 on windows - 2.04.93 is not available yet for the ConBee.
That means  I have to hold back till the new version on conbee.....

wvuyk, 2017-11-28:
deCONZ is compiled for windows, so this one can be closed

closed 2017-11-28 by wvuyk