dresden-elektronik/deconz-rest-plugin #299: Osram Lightify cla 60 rbgw - not parring

luci84tm, 2017-11-28:
Hi there,
I use a raspbee on my pi3 with deconz 2.04.92 in untended mode (systemd).
First time when I got my RGB Osram bulb it paired with no problem..
after that I turned the bulb off for a couple of days and when I tried to repair it never connected to my zigbee network..
Now deconz is detecting in settings the bulb as Light 1 and  but not showing up on the main page..
if i press reset it blinks shortly but that's it, if i click blink it blinks several times..
What can I do to get the device to  appear on the main page to be able to control it ?

manup, 2017-11-28:
For OSRAM bulbs the order is important, since after power-cycle they only search once for a network.

1) Reset the bulb is in factory new state
2) Open the network in the webapp
3) Turn bulb off and on again, it should then search and join the network

luci84tm, 2017-12-02:
Sorry for the late reply..
I followed your instructions several times but I can't make it join the network.
I tried manual reset 5sec on 5 off x5 times, then open network off and on, nothing...
sometimes it blinks once, almost believed that now something happened, but not the case..
I also tried the reset function in the webapp, it blinks once shortly in the same way like the manual reset, then I tried your procedure, nothing ..
any further ideas would be more then welcomed :)

manup, 2017-12-02:
Since it won't join, the network settings should be verified since they also can cause such problems. There is a FAQ entry which describes the relevant parameters. 

Please verify following parameters in deCONZ network settings:
* `Security Mode:`  High - No Master but TC Link Key
* `TC Address:`   Same as MAC address
* `TC Link Key:`   0x5a6967426565416c6c69616e63653039

Q: **How do I have to configurate the parameters in the deCONZ, that all lights and FLS-devices (also from other producers) can join my gateway/network?**

https://www.dresden-elektronik.de/funktechnik/service/support/faq/wireless-light-control/?L=1

luci84tm, 2017-12-02:
All this configs need to be done from the gui i guess ?

manup, 2017-12-03:
Yes, no API for that yet.

wrmulf, 2017-12-10:
I' m not sure whether my problem is related to your problem, but I moved yesterday from deconz stable (2.04.35) to the newest beta (2.04.94) and now my Osram Bulb "Classic A60 RGBW" renders unreachable after a while. Before it worked like a charm. When the bulb is power cycled, it is rendered active again, but sooner or later it is getting inactive again.

closed 2019-03-30 by stale[bot]