dresden-elektronik/deconz-rest-plugin #301: Conbee - web app throwing errors

wvuyk, 2017-11-29:
I have several users using the conbee on windows. A few users have issues approachng the webapp site where the following test is sown instead of the webapp:

**This is not the page you are looking for**
The file / couldn't be found.

This is often started when users want to restart deCONZ on a different port with --http-port=xx
Returning to the original port does not help then.
What steps can I do to solve this or create more info on this issue?

Thanks,

Wim

manup, 2017-11-29:
As alternative to using `--http-port=<port>` the default port can be changed in the config.ini.

On Windows it should be located in `C:\Users\<User>\AppData\Local\dresden-elektronik\deCONZ\config.ini`.

```
[http]
port=80
```

When no `--http-port=<port>` is specified the config.ini setting will be selected.

wvuyk, 2017-11-29:
We will test this, but why does specifying the parameter give an inaccessable issue?

Wim

wvuyk, 2017-11-29:
To be clear, the text

> This is not the page you are looking for
> The file / couldn't be found.

is only shown on the port that has been specified while starting deCONZ

manup, 2017-11-29:
Ah I see so the port is working. The error means that the webapp can't be located for some reason.

Normally it will be searched relative from the directory where deCONZ.exe is located: on my windows PC the path is `plugins\de_web`.

```
C:\Program Files (x86)\deCONZ\bin\deCONZ.exe
C:\Program Files (x86)\deCONZ\bin\plugins\de_web\
```

wvuyk, 2017-11-29:
the response that I am showing is by going to http:\<IP webapp>:port from a remote machine, or going to http:/localhost:port
Both do not work, for a installation based on defaults, so there is something going wrong after users decide to use a different port?

wvuyk, 2017-11-29:
Using the in file does not change the issue

manup, 2017-11-29:
The port seems not to be the problem since the error message is shown which wouldn't work if the port doesn't work.

What is the installation path?
Is `plugins\de_web\index.html` present?

wvuyk, 2017-11-29:
the index.htm is present in the deconz install (defaults, so it is the user directory here, not programs)

manup, 2017-11-29:
Ok I need to test it with user directory as install path

wvuyk, 2017-11-29:
yes, maybe that is of influence... i tried to run it with my clients installation using eleveated rights and also with standard rights... no difference. Inititial installation was working correctly, after a few devices were added, he dicovered the machine needed other software on port 80 and restarted deCONZ with port 8080. Since that the webapp was no longer reachable

Airey001, 2017-12-02:
So i have been greeting this back to version 86.  I fond the config file change a while ago didn't help me nor did the http port option i have to use

--http-port=8090 --ws-port=8091

to not get the error.
I use port 80 for local home website
HomeSeer uses port 8080

If i just called the exe with --http-port 8090 it was hit or miss it some calls it would get the error some not.  once i added the --ws-port from then on every start was good with those.  However i am getting this same error if i try to call the exe from a windows scheduled task.

Airey001, 2017-12-02:
I also just found i can force this behavior in windows on demand by trying to open the application from command line.

if i go to command and do this

Microsoft Windows [Version 10.0.14393]
(c) 2016 Microsoft Corporation. All rights reserved.

C:\Users\Administrator>"C:\Program Files (x86)\deCONZ\bin\deCONZ.exe" --http-port=8090 --ws-port=8091

or this

C:\Users\Administrator>cd "C:\Program Files (x86)\deCONZ\bin"

C:\Program Files (x86)\deCONZ\bin>deCONZ.exe --http-port=8090 --ws-port=8091

it always starts with the error

But if its a windows short cut with the paramaters it works with out issue.

manup, 2017-12-02:
I've found the bug, will be fixed in the next version.

Till then use the workaround:

`C:\Users\Administrator>"C:\Program Files (x86)\deCONZ\bin\deCONZ.exe" --http-port=8090 --ws-port=8091 --http-root="C:\Program Files (x86)\deCONZ\bin\plugins\de_web"`

closed 2019-03-30 by stale[bot]