dresden-elektronik/deconz-rest-plugin #303: Active icons in HASS blue after latest deCONZ (2.04.93)

furymountain, 2017-11-29:
Hi!

After the lastest beta update of deCONZ (2.04.93) the icon for the light entities from it shows as blue when active, instead of the themes “paper-item-icon-active-color” color as for all of the others. Downgrading deCONZ solves the problem for now, but have anyone else encountered this and knows why it happens?

[image]

Best regards!

euf0ria, 2017-11-29:
I have noticed this as well.

simonporter007, 2017-11-30:
I can't give an exact reason but I think the colour of light entities in hass is taken from the rgb_color attribute that is set. Presumably then it's setting a 0,0,255 rgb_color for some reason? You can check the states page to see I guess.

furymountain, 2017-11-30:
You are absolutely correct. When I switch the light on I get the "rgb_color" attribute which sets to 0,63,255. I set the "rgb_color" to 255,255,255 and the color is correct according to theme. It does however change it's status back to 0,63,255 after a few seconds again. I also got the effects/scenes "random" and "colorloop" which wasn't there before.

If I take a look in deCONZ webGUI it still looks like before, with only temperature settings.

For some reason deCONZ seems to handle these lights as RGB lights through the hue component in HASS, although they are only White lights with temperature on the 2.04.93 beta. In my case these are IKEA Trådfri White with temperature.

furymountain, 2017-11-30:
I did however get information that a deCONZ component is on it's way for HASS and hopefully in about two weeks for the next release. So I can probably live with the blue color until then if that solves the issue instead of the HUE platform. :)

simonporter007, 2017-11-30:
Again, for what it's worth, I've been testing that component. It works great so far :) I also get the RGB option under the new component but deconz isn't changing it like you're finding it.

furymountain, 2017-11-30:
Sounds really good. Then I won't bother with this issue under the HUE component. :) Thanks for the replies!

abmantis, 2017-11-30:
https://bitbucket.org/romanski/deconz-custom-module/
It that the component? Or another?

simonporter007, 2017-11-30:
https://github.com/Kane610/deconz

marthoc, 2017-12-10:
I have been testing @Kane610’s HA integration and can report that this blue light icon issue is still present. I think it’s caused by deCONZ treating these lights (in my case Hue white ambiance) as colour lights even though they do not have colour attributes.

ebaauw, 2017-12-10:
Yes, see #315.  All color attributes are 0; `xy` value `[0, 0]` translates to blue.  They show blue in HomeKit as well...

I was happy that I didn't take into account the `type` of the light when #288 surfaced.  Now I wish I had...

manup, 2017-12-10:
Working on a fix, should be available in the next version

furymountain, 2017-12-11:
Just installed deCONZ 2.04.97 beta and can confirm that the color attributes (ex. rgb_color) does no longer show for my IKEA White Spectrum bulbs which also results in correct color for the icons. Big thanks for the amazing support!

Closing this issue.

closed 2017-12-11 by furymountain