dresden-elektronik/deconz-rest-plugin #305: Philips Hue Motion Sensor not showing in the API

esseki, 2017-12-02:
Hi,

I can not find a way to make my Hue motion sensor show up in the API.
I can see it in Deconz, but when I call the "/api/xxx/sensors", I don't find the motion sensor.
I don't see either any group in the web app to assign lamps with the sensor.

I just would like to precise that I'm still using the web app and that I don't have access to the new beta app.

Thank you in advance for your support.

esseki, 2017-12-04:
Hi,

i realize my message was maybe not clear enough.
I have a Hue motion sensor and few Hue bulbs. My sensor was being used to switch on/off a bulb in a corridor.
Everything was working fine with the Hue gateway.

But I decided to move all my Hue bulbs to a conbee. Everything is working not that bad since then. My only problem is that I don't know how to reproduce the same use case in my corridor involving the motion sensor and a bulb.

As said in my original message it seems that the motion sensor is visible in deconz. But then I don't know how to assign the bulb in my corridor to it.

I also have a Hue switch and the assignment with a bulb is done with a group that is created and visible in the web app. I was expecting the same with the motion sensor.

Any help will be greatly appreciated :)

manup, 2017-12-04:
The beta just went public ;)

https://github.com/dresden-elektronik/phoscon-app-beta

Also please use latest deCONZ version 2.04.94.

In the Phoscon beta app you can add the hue motion sensor `Hamburger menu > Settings > Devices`. It should then create the related REST API resources.

[image]

To let the sensor control your lights, add a motion sensor control under `Edit > Sensor control`

[image]

And set the chechmark on the sensor control settings tab like this:

[image]

esseki, 2017-12-04:
Hi,

thank you for your quick response.
It worked like a charm :)

I'm really impressed by the quality of the Phoscon app and I'm loving more and more my Conbee !!!

esseki, 2017-12-05:
Reopening this issue as behaviour seems top be weird : 
- the bulb is switched on when motion is detected (expected behaviour)
- the bulb is switched off when no motion is detected (expected behaviour)
- quickly after the bulb is switched off, if i move in the corridor to switch it on again, it does not work.

I'm wondering if this is not due to the seetings i had to enter : 
[image]

As you can see what i'm trying to achieve is very simple : 
- if a motion is detected switch on the light, 
- if no motion is detected after 1 minute switch off the light

The problem is that it seems that after no motion has been detected for 1 minute, the sensor does not detect movements anymore for a while (maybe 1 or 2 minutes) as if the "Do nothing" rules set after 2 and 3 minutes in the screenshot above were preventing the sensor to detect a motion and to swith back the bulb on.

Is this what you meant by 
> beware this works but the interface is a bit cryptic at the moment..

Does it mean that this can not be achieved via the interface at this stage ?

manup, 2017-12-05:
Yes currently number of states is fixed and the end state must be the last one, this will be improved in the next days.

So the to get it to work the `Off` action should be put into the bottom state.

esseki, 2017-12-05:
Make sense.
A possible solution would be to allow duration smaller than a minute (down to the second).
It would allow having the 3 states occuring within a minute and more importantly, it would allow to dimm the light for less than a minute.
I don't want to have the light dimmed for 1 minute if it should be switched off after one minute. 
Whereas having the light on for 1 minute and the light dimmed for 15 more seconds would be nice !

manup, 2017-12-05:
Duration below a minute is difficult with most motion sensors since often they will sleep for that time, till sending the next impulse (Philips sensor can be faster here though).

The interface will allow to specify a simple setup with only two or three states like On and Off, which reflects your use case.

Note that from a people perspective 1 minute and or 15 seconds is very little time, in real world setups duration might be considered a bit more relaxed :)

esseki, 2017-12-05:
ok. I get your point.
i will wait impatiently for the next beta ;)

closed 2017-12-10 by esseki