dresden-elektronik/deconz-rest-plugin #306: EuControls support

SweNilsson, 2017-12-04:
I have just recieved a EuControls LCM-LV-ZB controller for my LED light driver (Hide-a-Lite). 
I can find the unit using rest-API but I cannot control it. 
Would it be possible to get support for that unit? It would be very appreciated!

manup, 2017-12-04:
Can you please provide more infos:
* screenshots from basic cluster
* screenshot endpoint list
* how does it look in the REST-API

Does it work when you control it directly in deCONZ on/off and level cluster?

SweNilsson, 2017-12-04:
I have no control of the unit at all. This is what I see.

[image]

manup, 2017-12-04:
The light isn't assigned to a group yet.

You need to press the `+` button and assign it to a group (for example the Tradrfi group).

SweNilsson, 2017-12-04:
Wow, sorry and thank you. 
That enables control, however I still cannot get the lights to lit. But that might be some other compatibility issue of course.

manup, 2017-12-04:
Since it's a 1–10 V control you may need to dimm up the light first in order to turn it on. For that press the three dots symbol next to the lights name and use the dimm bar.

SweNilsson, 2017-12-04:
I found the remaining issue. My driver was acting as a slave to another driver which generated.a conflict. De-connecting thr sync cable made it work! :)

manup, 2017-12-04:
Cool glad to hear that :)

closed 2018-01-12 by SweNilsson