dresden-elektronik/deconz-rest-plugin #308: bitron dimmer compatible?

MPIAE-IT, 2017-12-04:
Hey,
I'm using the RaspBee Gateway to control my IKEA Tradfri devices, which works perfect.
Now I need to connect some bulbs which are normal bulbs, which can't be exchanged to zigbee bulbs. Therefore I need a zigbee dimmer which is small. The dresden dimmers are too large. 

I want to use a Bitron Dimmer 
(http://www.bitronvideo.eu/index.php/produkte/smart-home-produkte/uaktor-dimmer/)
It communicates with ZigBee Home Automation Version 1.2.
Will this dimmer be usable with RaspBee and HomeKit?
If you need some data to make it compatible, I'm happy to buy one and give you all data you need.

Thanks

Lenny

manup, 2017-12-09:
It might work, since it is likely a normal on/off or dimmable light, but we haven't tested it yet.
If it doesn't work some screenshots from within deCONZ are helpful.

>Will this dimmer be usable with RaspBee and HomeKit?

If it works in deCONZ/REST API the `homebridge-hue` plugin can forward the device to HomeKit. I guess the device must be added to the plugins lights whitelist before.

MPIAE-IT, 2017-12-09:
Hey,
I just connected the bitronvideo AV2010/16 dimmer to my zigbee network. As far as i can tell, everything is working out of the box. 
I can control it through the deconz web interface as well as from HomeKit.

However I still wanted to upload some screenshots. If you need more, just tell me what I should send you.

Connecting the dimmer was a bit tricky, though. I don't know if I was doing something wrong or if the small function button just reacts to input very badly. Felt like it only recognised every 3rd or 4th input I gave. 

[image]

[image]

hpd96, 2017-12-18:
Hello,
using
"Bitron Home Smart Plug mit Schaltfunktion 16 A, weiß, 902010/28 "
http://www.bitronvideo.eu/index.php/produkte/smart-home-produkte/smart-plug-mit-schaltfunktion/
with deCONZ with succes.

BR/
Peter

closed 2018-11-23 by L3nny5