dresden-elektronik/deconz-rest-plugin #311: Sengled Element Bulbs

mercenaryballs, 2017-12-08:
I have these bulbs paired with my Conbee. I can see them in the GUI, but they do not show up in the web app. I believe they are unsupported. I can control them in the GUI (on/off and level control). Here are the atttributes: 

[image]
[image]
[image]
[image]
[image]
[image]

mysteryan, 2017-12-08:
I would try power cycling them.  If it doesnt discover properly, I would manually edit the session.default file and manually input working Extended and Simple Descriptors for that node.

manup, 2017-12-08:
Normally it should work since this seems to be a simple HA Dimmable Light with no special integration needed. Can you provide output by REST API?

mercenaryballs, 2017-12-08:
It does not show up in the Web App or the rest API. I have tried searching
for lights in the web app, but that seems to do the same thing as when I
open the network in the deconz gui. Here is the API output:

{
>
> "config":{
>
> "UTC": "2017-12-08T05:20:38",
>
> "apiversion": "1.0.4",
>
> "backup":{
>
> "errorcode": 0,
>
> "status": "idle"
>
> },
>
> "bridgeid": "00212EFFFF013EF7",
>
> "datastoreversion": "60",
>
> "devicename": "ConBee",
>
> "dhcp": true,
>
> "factorynew": false,
>
> "gateway": "192.168.10.1",
>
> "internetservices":{
>
> "remoteaccess": "disconnected"
>
> },
>
> "ipaddress": "192.168.10.15",
>
> "linkbutton": false,
>
> "localtime": "2017-12-08T00:20:38",
>
> "mac": "1c:1b:0d:9a:b0:ce",
>
> "modelid": "deCONZ",
>
> "name": "deCONZ-GW",
>
> "netmask": "255.255.255.0",
>
> "networkopenduration": 60,
>
> "panid": 38530,
>
> "portalconnection": "disconnected",
>
> "portalservices": false,
>
> "portalstate":{"communication": "disconnected", "incoming": false,
>> "outgoing": false, "signedon": false…},
>
> "proxyaddress": "none",
>
> "proxyport": 0,
>
> "replacesbridgeid": null,
>
> "starterkitid": "",
>
> "swupdate":{"checkforupdate": false, "devicetypes":{"bridge": false,
>> "lights":[], "sensors":[]…},
>
> "swupdate2":{"autoinstall":{"on": false, "updatetime": "" },
>> "bridge":{"lastinstall": "",…},
>
> "swversion": "2.4.94",
>
> "timeformat": "12h",
>
> "timezone": "none",
>
> "uuid": "0f333de3-b06e-4f65-ad63-7a0d23d5bf5e",
>
> "websocketnotifyall": true,
>
> "websocketport": 443,
>
> "whitelist":{
>
> "2C807EC599":{
>
> "create date": "2017-12-08T03:30:07",
>
> "last use date": "2017-12-08T05:20:28",
>
> "name": "deCONZ WebApp"
>
> },
>
> "988112a4e198cc1211":{
>
> "create date": "2017-12-08T04:31:48",
>
> "last use date": "2017-12-08T05:20:38",
>
> "name": "my application"
>
> },
>
> "F266FF0A54":{
>
> "create date": "2017-12-08T04:22:12",
>
> "last use date": "2017-12-08T05:14:29",
>
> "name": "Phoscon#B2560x900"
>
> }
>
> },
>
> "zigbeechannel": 15
>
> },
>
> "groups":{
>
> },
>
> "lights":{
>
> },
>
> "rules":{
>
> },
>
> "schedules":{
>
> },
>
> "sensors":{
>
> }
>
> }
>
>

Thanks,

Thomas Pendergrass
thomas.pendergrass@gmail.com
804-874-2570  Cell

manup, 2017-12-09:
You may retry with version 2.04.96, we have improved light detection there, it may help here.
After update I recommend to power cycle the lights to speed up discovery.

mercenaryballs, 2018-01-17:
I just got around to retrying this. Version 2.04.99 now. Still having the same issue. The lights are shown and can be controlled in the desktop app, but they are not shown in the web app or rest api.
[image]

`BODY
raw 
{"config":{"UTC":"2018-01-17T05:38:00","apiversion":"1.0.4","backup":{"errorcode":0,"status":"idle"},"bridgeid":"00212EFFFF013EF7","datastoreversion":"60","devicename":"ConBee","dhcp":true,"factorynew":false,"gateway":"192.168.10.1","internetservices":{"remoteaccess":"disconnected"},"ipaddress":"192.168.10.15","linkbutton":true,"localtime":"2018-01-17T00:38:00","mac":"1c:1b:0d:9a:b0:ce","modelid":"deCONZ","name":"deCONZ-GW","netmask":"255.255.255.0","networkopenduration":60,"panid":38530,"portalconnection":"disconnected","portalservices":false,"portalstate":{"communication":"disconnected","incoming":false,"outgoing":false,"signedon":false},"proxyaddress":"none","proxyport":0,"replacesbridgeid":null,"starterkitid":"","swupdate":{"checkforupdate":false,"devicetypes":{"bridge":false,"lights":[],"sensors":[]},"notify":false,"text":"","updatestate":0,"url":""},"swupdate2":{"autoinstall":{"on":false,"updatetime":""},"bridge":{"lastinstall":"","state":"noupdates"},"checkforupdate":false,"install":false,"lastchange":"","lastinstall":"","state":"noupdates"},"swversion":"2.4.99","timeformat":"12h","timezone":"none","uuid":"0f333de3-b06e-4f65-ad63-7a0d23d5bf5e","websocketnotifyall":true,"websocketport":443,"whitelist":{"2C807EC599":{"create date":"2017-12-08T03:30:07","last use date":"2018-01-17T05:37:31","name":"deCONZ WebApp"},"8F61CDD11E":{"create date":"2018-01-17T05:37:38","last use date":"2018-01-17T05:38:00","name":"my application"},"988112a4e198cc1211":{"create date":"2017-12-08T04:31:48","last use date":"2017-12-08T05:20:38","name":"my application"},"F266FF0A54":{"create date":"2017-12-08T04:22:12","last use date":"2017-12-09T06:46:28","name":"Phoscon#B2560x900"}},"zigbeechannel":15},"groups":{},"lights":{},"rules":{},"schedules":{},"sensors":{"1":{"config":{"on":true,"reachable":true},"ep":1,"etag":"a223b40012aa853f966daafeb7884cf0","manufacturername":"LUMI","modelid":"lumi.sensor_magnet.aq2","name":"OpenClose 1","state":{"lastupdated":"2017-12-22T06:11:08","open":false},"type":"ZHAOpenClose","uniqueid":"00:15:8d:00:01:e4:4d:ae-01-0006"}}}`

staraxis, 2018-01-31:
I have the same issue with a Sengled bulb as well. Shows up and can be controlled via the GUI but not available in the rest API or the webapp. Any workaround or update on this?

ebaauw, 2018-01-31:
Looks like another end-device light.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/356#issuecomment-358759456.

staraxis, 2018-01-31:
Thanks @ebaauw , is there an update on the way to fix this?

ebaauw, 2018-01-31:
You'll have to ask @manup.  I don't feel comfortable messing with this part of the code.

manup, 2018-02-01:
Hi I've put it on the todo list should be solved soon, as noted currently only router lights will be recognized by the API.

staraxis, 2018-02-09:
Any update on the addition of end-device lights to the API and/or webapp? Thanks

ebaauw, 2018-02-11:
I see @manup included my PR in v2.05.02.  I think they should be recognised by that version, but cannot test myself as I don't have any end-device light or plug.

staraxis, 2018-02-12:
Thanks @ebaauw and @manup I can confirm that my sengled light is now available via the webapp and also in my Home Assistant UI after updating to v2.05.02. On/Off and brightness work as expected. Cheers

mercenaryballs, 2018-02-15:
My bulbs are working as well. They show up on the phoscon app and in the REST API.

closed 2018-02-15 by mercenaryballs