dresden-elektronik/deconz-rest-plugin #313: Problems adding some devices

rtenklooster, 2017-12-08:
My current setup (build 96) is having some problems with adding new (XIAOMI) devices.
Currently I have about 22 devices connected. 3 Ikea bulbs. A lot of Motion sensors, open close sensors, switches etc.

The problem I'm facing is beeïng unable to add a couple of door sensors.
Two of my PIR's stopped working. What I did was this:
Made a backup, reset my raspbee to factory settings using the webapp.
After opening the network, I'm able to add the 3 door sensors and 1 switch (Aqara) to the "fresh" network.
However, when I restore my backup, I want to try to re-add those problem sensors to the "old" network. Just nothing happens.

I can add them to the Xiaomi gateway without any problems.

So they sensors are compatible to deconz, they are working, I just can't add them to my current network.
I thought, well let's just use the new one and add all my sensors.. however after 11 nodes, I was unable to add new sensors anymore.

Any suggestions on what could be the problem, of how to resolve this?

bdickhaus, 2017-12-09:
I am having a similar problem adding Xiaomi devices to my network. I have four Hue sensors, four Hue switches and 14 Hue lights. I have tried all kinds of different ways of adding the Xiaomi devices. They do show up in deCONZ, but they have only one circle (no clusters) and either no dot, or a black dot. I am able to get them to link to the bridge with a little effort, but they still are not recognized. Running build 94 and 86 (I am currently back on 86). I was going to try what rtenklooster did, but now I don't have to.

manup, 2017-12-09:
The door sensors are tricky to add, in our tests the best approach was:
* *Devices > Sensors > Add other sensor* (Phoscon App)
* press the sensor button for > 5 seconds until LED lit up
* open/close the sensor contact many times so that it sends commands and is recoknized

manup, 2017-12-09:
>I thought, well let's just use the new one and add all my sensors.. however after 11 nodes, I was unable to add new sensors anymore.

Sensors always need a parent node to communicate through (gateways or light), since you just have 3 lights maybe you reached child limit here, not sure how many end devices the RaspBee support directly, have to check ... Adding ore mains powered devices may help here.

rtenklooster, 2017-12-10:
Did a new check. 
I'm only connecting door/windows sensors (no meshing etc.). Changed the channel, erased the db, reset the gateway so everything is new. 

Again: after adding 10 sensors, the 11th won't pair. I'm sure it's either a bug, or a limit somewhere. 
The previous try also gave problems when 10 child's where connected. 

All sensors where added using the new webapp, everyone without even a retry, just click, and succesfull added. Untill 10 devices, deCONZ shows 11 nodes, including the GW.

manup, 2017-12-10:
I can confirm this, just checked the related firmware limits configuration.

Exactly 10 end devices: so to support more sensors in the network other mains powered devices like lights must be added.

I think in a future firmware the limit can be raised but needs more testing on RAM usage.

[image]

rtenklooster, 2017-12-10:
Ok cool :-) 
I have a couple of routers laying around so no problems for now. Will give it a try right now and will post the results.

rtenklooster, 2017-12-10:
Adding a MI outlet and voilla:

[image]

rtenklooster, 2017-12-11:
Thanks for looking into the souce code. When I have to test with a modified max, please let me know!

closed 2017-12-11 by rtenklooster

Cavemanz, 2018-06-25:
I ran into the exact same problem. After 10 devices I couldnt add any more no matter what I tried (restarting DeCONZ docker, Ubuntu VM, NUC ESXi host). After adding a Philips Hue Go I had spare, I could add more devices. This should be part of the supported devices wiki or something, so people understand theres a limit to how many battery sensors you can add before hitting this limit.