dresden-elektronik/deconz-rest-plugin #314: Xiaomi switches do not send >quadruple (4) clicks

ensamresande, 2017-12-09:
I noticed recently that deconz support quadruple-clicks for Xiaomi switches (and possible others). Nice!

However it seems that if you press more than 4 times, no event is fired at all. This is rather irritating since you sometimes accidentally press an additional time(s) while quickly wanting to do a quadruple-click.

How about changing to some dynamic n-click support where any number of clicks is supported? That way I can translate >= 4 to 4 further down the chain.

manup, 2017-12-09:
I'm afraid it might not be possible, the switch does always send just one command after a short delay (< 1 second) in which the button press count is a parameter. Not sure if it supports more than 4 clicks.

ensamresande, 2017-12-09:
Oh, so it is the switch which does the logic for the multiple clicks? I thought this was handled in deconz. :(

And I'm guessing whatever count parameter the switch sends is just forwarded by deconz to (in my case) the websocket?

manup, 2017-12-09:
> Oh, so it is the switch which does the logic for the multiple clicks? I thought this was handled in deconz. :(

Yes that's handled by the switch, only one command is send, good for battery.

> And I'm guessing whatever count parameter the switch sends is just forwarded by deconz to (in my case) the websocket?

That's correct, just testet it when pressed more than 4 times the switch sends nothing.

ensamresande, 2017-12-09:
Booo! :(

closed 2017-12-09 by ensamresande