dresden-elektronik/deconz-rest-plugin #318: Changing color temperature with Trådfri remote

gbroll, 2017-12-10:
This is more of a feature request than an issue, but I'm don't know where else I could post it...

I'm running deconz 2.04.94 on a rpi3 with conbee, approx ~10 trådfri lights and a few remotes. Works like a charm, but I was wondering if it would be possible to enable changing the color temperature "continuously" with (long) press on the remote buttons (i.e. buttonevent 4001, 5001), just like dimming works (buttonevents 2001, 3001). Currently, it appears that the color temperature can only be changed with short clicks (4002,5002).  Or have I missed something?

bofax, 2017-12-11:
Im just curious.. do you use TRADFRI-Remotes? How did you get to integrate them? With the Web-Api?

gbroll, 2017-12-11:
Yes, just reset them to factory settings and press the on/off button (perhaps a couple of times), and they show up in deconz GUI as well as in the webapp.

manup, 2017-12-11:
Also, to setup cold/warm white control after pairing, press the large button for about 10 seconds.

chr55, 2017-12-11:
I would like to follow this procedure with my Tradfri remote controls so that I can control the cold/warm settings with the left/right buttons. How do I remove the remote from deCONZ before resetting the remote to factory default? Or, is this not necessary, but wouldn't I get a double in deCONZ?

thanks,

simonporter007, 2017-12-17:
I replicated this outside of deconz with repeated calls to deconz API with transitiontime but it's not as smooth as the default remote to bulb behaviour. +1 if this would be possible in deconz itself

closed 2018-09-17 by manup