dresden-elektronik/deconz-rest-plugin #319: Any chance supporting Salus TRV10RFM?

markuspf, 2017-12-11:
I am looking at getting these (or, for that matter any) working TRV. Any chance of them getting supported?

kylegordon, 2018-12-13:
I am also interested in TRV support, such as for the TRV10RFM above, or the eCozy 1TST-EU. There has been some discussion at https://github.com/dresden-elektronik/deconz-rest-plugin/issues/209

manup, 2018-12-13:
A few weeks ago I've tried to join the TRV10RFM to deCONZ while sniffer was running, after many attempts it did, but then left the network, likely since it waited for some configuration.

So more work needs to be done to inspect what the original gateway UG600 configures, unfortunately I don't have this gateway.

kylegordon, 2018-12-13:
@markuspf Do you have the Salus UG600 gateway?

markuspf, 2018-12-20:
@kylegordon no, also I doubt I'd have the time to do any sniffing/debugging. The Gateway itself seems ludicrously expensive (£95 on amazon) as well...

mindvisionro, 2019-01-06:
Hello, i have the UG600, i want to integrate with home assistant, but for now no results.
If i can help you with some details let me know.

bencollins54, 2019-01-19:
@manup Any way of making progress with this without the UG600 ?

closed 2019-05-26 by stale[bot]