dresden-elektronik/deconz-rest-plugin #321: Determine Current Firmware

chr55, 2017-12-11:
I am trying to find out what firmware version my ConBee has. Can anyone point me in the right direction?  I am using the Raspberry Pi image updated to 2.04.93.

I see files on the sd card showing 26190500 but I can't find out where to see what my ConBee is actually running with. 

thanks,

chr55, 2017-12-11:
Thank you. I did not know I could VNC into the Raspberry Pi version, I thought the web ui was all that was available. I see that I am at version 0x26190500 by using Help / About in VNC deCONZ.

chr55, 2017-12-11:
My current ConBee was being run by version 2.04.93 Raspberry Pi. When I tried 2.04.97 the group names were all defaulted back to as an example "Group 1". I can edit those back to what they are supposed to be which would not be a problem. 

What may be a problem for me is that all the remote controls did not show in 2.04.97 web ui under Devices. I had three hue remote controls showing in 2.04.93. I also have several IKEA remote control devices which never showed under Devices in 2.04.93 but were working ok.

Do I delete the remote controls and add them back into 97?

thanks,

manup, 2017-12-11:
Did you install the new sd-card image or change the host?

Stuff like group names and paired sensors are stored in the file `zll.db`, so when installing a new sd-card image or move ConBee to a different device (Raspberry or PC) this file must be copied as well.

The best is to copy the whole folder `/home/pi/.local/share/data/dresden-elektronik`.

chr55, 2017-12-11:
That procedure worked great. Thank you.

On the old Raspberry Pi .93 I ran
sudo systemctl stop deconz
Then connected via VNC and used the file explorer to copy the directory you indicated to my local network somewhere (/home/pi/.local/share/data/dresden-elektronik).

I shutdown .93 Raspberry Pi
Inserted new SD card with fresh version .97 on it and booted up
sudo systemctl stop deconz
Connected via VNC to .97 and copied the directory back from the network to the local filesystem (/home/pi/.local/share/data/dresden-elektronik)
sudo systemctl start deconz

Everything came back like it was including group names and the remote control devices.

thank you,

closed 2017-12-11 by chr55