dresden-elektronik/deconz-rest-plugin #322: Issues with OSRAM lights

wrmulf, 2017-12-12:
Since I started evaluating deCONZ-Beta I discovered many problems with my OSRAM lights. After a few hours they are rendered offline/unreachable und will only rejoin the zigbee network after power cycling the lights. 

The following lights show the issue:
- one of my "Classic B40 TW - LIGHTIFY" (E14, "swversion": "V1.03.20") (one of three)
- one "Classic A60 RGBW" (E27, "swversion": "V1.03.07") (the only one I own)
- one "Flex RGBW" (LED stripe, "swversion": "V1.03.07") (the only one I own)

Not affected are:
- OSRAM "Plug 01"  ("swversion": "V1.04.12")
- OSRAM "Surface Light W23 LIGHTIFY" ("swversion": "V1.03.20")
- one Philips "LWB006" ("swversion": "5.38.2.19136")

I never had any problems with deCONZ stable, but I was interested in new features (Alexa compatibilty, better API compatibility to HUE, etc.). Curiosity killed the lights.

Best regards,
Robert

easybeat, 2018-03-10:
I have same issues with my Osram GU10 bulbs firmware 1.03.21.

My bulbs do not react after a first command and I have to power cycling them and then they work again 1x!

Thank you

Beat

roflmao, 2018-03-10:
I have issues with my OSRAM bulbs with old firmware as well #447. Someone have access to firmware let me know :) They work well with old firmware on my Hue hub though.

jonathanweinberg, 2018-05-05:
The SmartThings Gen 2 Hub can update Osram bulbs. I wonder if I can snoop the traffic to see where the firmware comes from.

closed 2019-03-30 by stale[bot]