dresden-elektronik/deconz-rest-plugin #327: Tradfri color bulb is switching to nonexistent colormode "ct"

mariusmotea, 2017-12-14:
Hi,

I discover one issue when i try to apply last know state to all lights. When switched off/on the Tradrfri color bulb is switched to colormode "ct". Issue is that this bulb don't support color temperature and has no ct state field and this is causing me errors because the value is missing for specified colormode.

Marius.

ebaauw, 2017-12-14:
See #218 item 4.

tpetri, 2018-01-06:
I'm a new conbee user, I have now tried to follow the discussions here, and it seems that for my two "TRADFRI bulb E27 WS opal 980lm" bulbs the scene store/recall does not work correctly since the colormode is set to "xy" instead of "ct" after a store/recall cycle and of course the color of the bulb is wrong after recall, this seem to have been resolved now several times in the past I thought, what is the current state on this, is this a regression or is the bulb not covered by the special handling for some IKEA lamps?

ebaauw, 2018-01-06:
I don’t think this has been resolved yet or in the past.  Only as a workaround I was able to store a scene correctly using the deCONZ GUI, but not the REST API.

tpetri, 2018-01-07:
Thanks for the hint, I tried now setting the color with x/y values in deCONZ directly and then storing the scene - this works like a charm, it recalls the correct color setting.
Is there a way in the deCONZ GUI to set the color mode manually? Then I could do some more testing. Currently it seems that scene storing is hardcoded to colormode x/y in the IKEA bulbs somehow. But ok, for me this is a workaround, super annoying, but it works.

ebaauw, 2018-01-07:
> Is there a way in the deCONZ GUI to set the color mode manually?

No, in ZigBee, Colormode is a read-only attribute (or actually two read-only attributes: Colormode and Enhanced Colormode - that's part of causing all this nonsense).  You cannot set these - the light's firmware **should** (that's the other part) set these, according to the last 0x0300 command used to change the colour: based on X/Y, based on (Enhanced) Hue/Saturation, or based on Color Temperature.

The IKEA lights are odd, in that a) the colour bulb doesn't support Hue/Saturation, and b) the colour and white spectrum lights advertise both Colormode and Enhanced Colormode, but only update one of these.  a) should shouldn't be an issue, as the light does advertise it doesn't support Hue/Saturation, but deCONZ doesn't use that info; b) is an issue and can only be worked around by whitelisting the light.

closed 2019-03-30 by stale[bot]