dresden-elektronik/deconz-rest-plugin #328: Emulate unsupported hardware features in Software

Hypfer, 2017-12-14:
Apparently, the IKEA Tradfri RGB Bulb doesn't support anything besides CIE-xy. Neither HSV nor the color loop feature works. Sending these commands just fails silently.

Both features should be easy to emulate in Software so I'd suggest implementing a completely transparent shim to do exactly that.

ebaauw, 2017-12-14:
> Sending these commands just fails silently.

That's a bug.  At least, deCONZ should not expose these attributes for the IKEA bulb, see #218.

> Both features should be easy to emulate in Software

Not really.  Colorloop cannot be implemented on the gateway - it needs to be supported by the bulb.  Currently, deCONZ translates `hue`/`sat` to the ZigBee command to change _Hue_ and _Saturation_ (which the IKEA bulb doesn't support) and `xy` to the ZigBee command to change _CurrentX_ and _CurrentY_.  It would be possible to convert `hue` and `sat` from the API to the `xy` values in the gateway (I do this in homebridge-hue, as HomeKit only provides RGB hue and saturation), but it's not a trivial change, see https://developers.meethue.com/documentation/color-conversions-rgb-xy.  Also note this would break compatibility with the Hue bridge or with lights supporting _Hue_ and _Saturation_, where these values are _not_ in the RGB colour space, but dependent on the light's supported colours.

Hypfer, 2017-12-17:
Why can't colorloop be implemented on the Gateway?
```
Send set color A
sleep a few seconds
Send set color B
etc.
```

manup, 2018-09-17:
Would cause a lot of ZigBee traffic :)
Closing the oldest issues for know to tidy up the tracker and duplicates in newer issues.

closed 2018-09-17 by manup