dresden-elektronik/deconz-rest-plugin #329: Phoscon app doesn't work on Raspberry Stretch default web browser

ebaauw, 2017-12-14:
After one of the recent `apt upgrade` runs, Stretch now provides GNOME Web 3.8.2 as default web browser, see https://wiki.gnome.org/Apps/Web.  Somehow that browser doesn't render Phoscon (as distributed with v2.04.97) correctly.  The old web app seems to work under Web; copying/pasting the Phoscon URL into Chromium (which is still installed) also works.

manup, 2017-12-15:
Yes currently the Phoscon App works and is testet with Chrome, Firefox, Safari, Samsung Browser and Edge (early support). I think on Raspberry the Phoscon App button should open the chromium based browser and maybe issue a warning if opened in a unsupported browser (or when certain features are missing).

marthoc, 2017-12-15:
Phoscon works great on Safari for Mac but the layouts are a little messed up on mobile Safari.

manup, 2017-12-16:
> Phoscon works great on Safari for Mac but the layouts are a little messed up on mobile Safari.
On which device and iOS version is your mobile Safari?

We do test with Safari 11 on desktop and Safari 11 iPad (2016), Safari 10 iPad + iPhone 5.

If you can provide a screenshot we check what the issue is.

marthoc, 2017-12-16:
iPhone 7 Plus on iOS 11. Here are some screenshots:
[image]
[image]
[image]
[image]
On the sensor and switch pages I’m missing the “Add” button and an extra Column of information unless I turn my device sideways. Then I can see the Add button. But if I click it then you can also see the next screen for selecting the vendor is not scaling properly.

manup, 2017-12-16:
Ah I see, yes indeed the mobile phone sensor and switch wizards as well as phone landscape mode are not finished yet. The for now wizards require at least tablet size screen.

marthoc, 2017-12-17:
Ok! I can live with the layouts because the functionality works as it should! :)

closed 2018-06-12 by ebaauw